open-logic/open-logic #41: Signal 'M_Axi_RResp' is read in the process but is not in the sensitivity list

kuriousd, 2024-07-23:
After synthesis of `olo_axi_master_simple.vhd` using Vivado, I got the following warning:

> [Synth 8-614] signal 'M_Axi_RResp' is read in the process but is not in the sensitivity list ["/home/fpgauser/vivado-projects/aes-256/open-logic/src/axi/vhdl/olo_axi_master_simple.vhd":244]

Regards
Fer.

obruendl, 2024-07-23:
Thank you for reporting - reporting issues (so I can fix them) is a key benefit of the community.

Fitst of all: the core still should work, the issue should not affect fuctionality on HW.

I will fix the issue on the develop branch. Would it be acceptable for you to clone develop for the next few weeks (until the next release) or do you need a release/tag urgently?

Every release adds some overhead, so I try avoiding releases for fixing a single bug.

kuriousd, 2024-07-23:
There’s no problem on my side. Just reporting a warning which is easy to solve.

I keep pointing to 2.0.0 release

obruendl, 2024-07-23:
@kuriousd 

I fixed it on *develop*. The fix will be  released in the 2.1.0 release.

Thank you for reporting.

obruendl, 2024-07-23:
@kuriousd 

BTW: I appreciate if you share your experience with Open Logic on linked-in. Of course I am also very open for any DMs regarding suggestions for improvements.

obruendl, 2024-07-30:
Closed with release 2.1.0

closed 2024-07-30 by obruendl