open-logic/open-logic #43: VSG Linting

obruendl, 2024-07-24:
@patrick-studer implements VSG linting.

The issue is meant to discuss linting and linting-rule related questions

patrick-studer, 2024-07-25:
## FSM- and Enum Types

### Observation

#### FSM
- \src\axi\vhdl\olo_axi_lite_slave.vhd (1 hit)
  - `type Fsm_t is (Idle, WrCmd, WrData, WrResp, RdCmd, RdData, RdResp);`
- \src\axi\vhdl\olo_axi_master_full.vhd (5 hits)
  - `type WriteCmdFsm_t is (Idle_s, Apply_s);`
  - `type WriteWconvFsm_t is (Idle_s, Transfer_s);`
  - `type WriteAlignFsm_t is (Idle_s, Transfer_s, Last_s);`
  - `type ReadCmdFsm_t is (Idle_s, Apply_s, WaitDataFsm_s);`
  - `type ReadDataFsm_t is (Idle_s, Transfer_s, Wait_s);`
- \src\axi\vhdl\olo_axi_master_simple.vhd (5 hits)
  - `type WriteTfGen_s is (Idle_s, MaxCalc_s, GenTf_s, WriteTf_s);`
  - `type ReadTfGen_s is (Idle_s, MaxCalc_s, GenTf_s, WriteTf_s);`
  - `type AwFsm_s is (Idle_s, Wait_s);`
  - `type ArFsm_s is (Idle_s, Wait_s);`
  - `type WFsm_s is (Idle_s, NonLast_s, Last_s);`
- \src\intf\vhdl\olo_intf_i2c_master.vhd (1 hit)
  - `type Fsm_t is ( BusIdle_s, BusBusy_s, MinIdle_s, Start1_s, Start2_s, WaitCmd_s, WaitLowCenter_s,`
- \src\intf\vhdl\olo_intf_spi_master.vhd (1 hit)
  - `type State_t is (Idle_s, SftComp_s, ClkInact_s, ClkAct_s, CsHigh_s);`

#### ENUM
- none

#### ARRAY
- \src\base\vhdl\olo_base_arb_prio.vhd
  - type `Data_t is array (natural range<>) of std_logic_vector(Out_Grant'range);`
- \src\base\vhdl\olo_base_cc_pulse.vhd
  - type `Pulse_t is array (natural range <>) of std_logic_vector(NumPulses_g - 1 downto 0);`
- \src\base\vhdl\olo_base_delay.vhd
  - type `Srl_t is array (0 to MemTaps_c - 1) of std_logic_vector(Width_g - 1 downto 0);`
- \src\base\vhdl\olo_base_delay_cfg.vhd
  - type `Srl_t is array (0 to 1) of std_logic_vector(Width_g - 1 downto 0);`
- \src\base\vhdl\olo_base_dyn_sft.vhd
  - type `Data_t is array (natural range <>) of std_logic_vector(In_Data'range);`
  - type `Shift_t is array (natural range <>) of std_logic_vector(In_Shift'range);`
- \src\base\vhdl\olo_base_pkg_array.vhd
  - type `t_aslv2    is array (natural range <>) of std_logic_vector( 1 downto 0);`
  - type `t_aslv3    is array (natural range <>) of std_logic_vector( 2 downto 0);`
  - ...
  - type `t_ainteger is array (natural range <>) of integer;`
  - type `t_areal is array (natural range <>) of real;`
  - type `t_abool is array (natural range <>) of boolean;`
- \src\base\vhdl\olo_base_pkg_logic.vhd
  - type `StageOut_t is array (natural range <>) of std_logic_vector(Pwr2Width_c - 1 downto 0);`
- \src\base\vhdl\olo_base_pl_stage.vhd
  - type `Data_t is array (natural range <>) of std_logic_vector(Width_g - 1 downto 0);`
- \src\base\vhdl\olo_base_ram_sdp.vhd
  - type `data_t is array (natural range<>) of std_logic_vector(Width_g - 1 downto 0);`
- \src\base\vhdl\olo_base_ram_sp.vhd
  - type `data_t is array (natural range<>) of std_logic_vector(Width_g - 1 downto 0);`
- \src\base\vhdl\olo_base_ram_tdp.vhd
  - type `data_t is array (natural range<>) of std_logic_vector(Width_g - 1 downto 0);`
- \src\intf\vhdl\olo_intf_debounce.vhd
  - type `Cnt_a is array (0 to Bits_c-1) of Cnt_t;`

I guess you see the inconsistency here...

### Proposal
- FSM
  - Keep for all types always `_t` since it is holding the type of the state holding signal and not a state itself.
  - Use the wording `XyzFsm_t` in the type definition.
  - Use `_s` for the state definitions (elements).
  - Use the wording `XyzState` for the signal of type `XyzFsm_t`.
  - Example:
    - `type WriteTfGenFsm_t is (Idle_s, MaxCalc_s, GenTf_s, WriteTf_s);` 
    - `signal WriteTfGenState :  WriteTfGenFsm_t := Idle_s` 
- ENUM
  - Keep for all types always `_t`.
  - Remove any suffix in enum-like definitions (elements) not representing a FSM state.
  - Example:
    - `type BackgroundColor_t is (Red, Green, Blue);` 
- ARRAY
  - Keep for all types always `_t`.
  - Use the wording `ElementNameArr_t` in the type definition .
  - Example:
    - `type InDataArr_t is array (natural range <>) of std_logic_vector(InData'range);`
    - `type `Slv3Arr_t is array (natural range <>) of std_logic_vector( 2 downto 0);`

@obruendl do you agree with this proposal?

patrick-studer, 2024-07-25:
##  Constants
Constants are currently differently handled depending on where they are (packages, functions/procedures, or architectures).
Sometimes even inconsistently in one domain.

### Observation

#### Package
- \src\axi\vhdl\olo_axi_pkg_protocol.vhd
  - `constant xRESP_EXOKAY_c : Resp_t  :=  "01";`
  - `constant xRESP_SLVERR_c : Resp_t  :=  "10";`
  - `constant xBURST_FIXED_c : Burst_t :=  "00";`
  - `constant xBURST_WRAP_c  : Burst_t :=  "10";`
  - `constant AxSIZE_1_c     : Size_t  := "000";`
  - `constant AxSIZE_2_c     : Size_t  := "001";`

#### Functions + Procedures
- \src\base\vhdl\olo_base_pkg_logic.vhd
  - `constant c           : std_logic_vector(size - 1 downto 0) 	  := (others => '0');`
  - `constant argDt       : std_logic_vector(arg'high downto arg'low) := arg;`
  - `constant Stages_c    : integer := log2ceil(inp'length);`
  - `constant Pwr2Width_c : integer := 2**Stages_c;`

### Proposal
- Not making a difference where constants are used, they always are PascalCase (maybe with underscores to separate groups/interfaces) and have the `_c` suffix.
- Example:
  - `constant Axi_RespExOkay_c : Axi_Resp_t  :=  "01";`
  - `constant Axi_RespSlvErr_c : Axi_Resp_t  :=  "10";`
  -  `constant ReturnVal_c  : std_logic_vector(size - 1 downto 0)  := (others => '0');`
  -  `constant ArgDt_c  : std_logic_vector(arg'high downto arg'low) := Arg;`

@obruendl do you agree with this proposal?

patrick-studer, 2024-07-26:
## Attributes

What should be the case-check for attribute definitions?

## Observation
- \src\intf\vhdl\olo_intf_sync.vhd
  - `attribute syn_srlstyle  : string;`
  - `attribute shreg_extract : string;`
  - `attribute ASYNC_REG     : string;`
  - `attribute dont_merge    : boolean;`
  - `attribute preserve      : boolean;`

### Proposal
I would follow the Xilinx style and go to upper-case.
- Example:
  - `attribute SYN_SRLSTYLE  : string;`
  - `attribute SHREG_EXTRACT : string;`
  - `attribute ASYNC_REG     : string;`
  - `attribute DONT_MERGE    : boolean;`
  - `attribute PRESERVE      : boolean;`

@obruendl do you agree with this proposal?

patrick-studer, 2024-09-06:
@obruendl 

Could you please provide feedback to the three open "Proposals" from above.

obruendl, 2024-09-12:
@patrick-studer 
Fsm and Enum types: agreed

obruendl, 2024-09-12:
@patrick-studer 
For attributes i prefer lower case. Reasons: 
- I also use lowercase for built-in attributes (SomeVector'high)
- I have a strong dislike for any ALL_CAPITAL identifiers. It's bad to hold shift all the time and it is bad for readability because it draws all attention on it.
- I don't care about AMD style because the library is vendor independent and there is no consent between vendors (altera & efinix seem to favor lowercase attributes). Hence I prefer following my personal preference.

obruendl, 2024-09-13:
@patrick-studer 
Could you add the definitions we agreed on to the Conventions.md file?

obruendl, 2024-10-14:
@patrick-studer 

It is done. VSG Linting is up and running, documented, integrated into CI and also into VSCode. Thank you so much for your support on this topic.

I have a few questions regarding VSG I would be very glad if you could investigate - so the setup could get even cleaner:
1. VSG seems unable to handle procedure without arguments - indentation was messed up in this case. I worked around this by adding an ugly "dummy" parameter. Could you check if it can be configured to accept this case? Probably it is a problem with the indentation rules. Example [here](https://github.com/open-logic/open-logic/blob/7b0ef64b3a840b31c8be15494833fc0903069acb/test/tb/olo_test_i2c_vc.vhd#L458)
2. Is it possible to enforce snail_case for VC procedure arguments? I configured them for "case: lower" - but this does not report errors if I name an argument in camelCase (starting lower-case but containing upper-case letters).
3. Range declarations for unconstrained record elements are off by one column if spread over multiple lines (otherwise VSG does report errors). Can this be fixed? Example [here](https://github.com/open-logic/open-logic/blob/7b0ef64b3a840b31c8be15494833fc0903069acb/test/axi/olo_axi_master_full/olo_axi_master_full_tb.vhd#L66)
4. I could upgrade VSG to 3.27 without much trouble. However, beyond that (I believe we are at 3.9.0 right now) the configuration fails. Is there any way to upgrade to newer VSG versions without completely re-engineer the config? I would like to not stay too far behind - this somewhen becomes a maintenance problem. This question is less about "how to upgrade to 3.9.0" than about "how to upgrade to newer versions in general"

obruendl, 2024-10-31:
Point 1 was (partially resolved). I had to change the indentation scheme a bit to avoid having to use a dummy - but the changes are acceptable and I prefer them over having the dummy arguments. Hence this is regarded as resolved.

obruendl, 2024-10-31:
Point 3 is a confirmed VSG bug: https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/1250

However, for open-logic this is regarded a minor issue - especially since it was confirmed that this is not related to the open-logic VSG config. Hence this is not followed up on.

obruendl, 2024-10-31:
Point 2 is still open:

https://github.com/jeremiah-c-leary/vhdl-style-guide/issues/1249

obruendl, 2024-11-04:
Issue closed.

All points that can be achieved with current VSG 3.27 are implemented. For the other topics PRs are filed in VSG and confirmed to fix the known issues. They will be incorporated after the next VSG release - this is noted in the feature ideas in the wiki.

closed 2024-11-04 by obruendl