open-logic/open-logic #50: AXI awid, bid, arid, rid signals missing in olo_axi_master_simple and olo_axi_master_full

kuriousd, 2024-08-02:
Hi,

After integrating olo_axi_master_simple in one of my designs and make it work in a board. I want to go a step further, integrate the design serving as interface for one of my IPs. I follow a cocotb approach to test my cores. cocotb presents a python package to easily implement verification components for AXI (https://github.com/alexforencich/cocotbext-axi).

If I use the AXI interface provided in olo_axi_master_simple the compilation in cocotb fails, because it consider the signals awid, bid, arid and rid mandatory [cocotb axi_channels.py](https://github.com/alexforencich/cocotbext-axi/blob/master/cocotbext/axi/axi_channels.py#L28) . I just add them to my ip-core and put them to (others =>'0') for the moment.

Maybe it could be interesting for the cores to add these four signal interfaces.

Regards,
Fer.

obruendl, 2024-08-03:
@kuriousd 

I omited any signals that are not used inside the *olo_axi_mater_simple/full* on purpose. I don't want to blow up the entity declaration and provide parameters to set the width of those signals if they are not used anywhere.

The idea is that if tools require those signals, the user can add them in his entity (the one that instantiates *olo_axi_master_simple/full*). As far as I understand this is exactly what you did - so you follow the path that I foresee.

Two questions:
1. Do I overlook anything (i.e. is the approach of adding them in your entity somehow problematic)?
2. Because you contacted me I checked and obviously my decision is not well documented. Would you mind extending the documentation and handing in a pull request?
  * https://github.com/open-logic/open-logic/blob/main/doc/axi/olo_axi_master_simple.md
  * https://github.com/open-logic/open-logic/blob/main/doc/axi/olo_axi_master_full.md

Of course you will be mentioned as contributor then.

kuriousd, 2024-08-05:
Sure, I can extend it

obruendl, 2024-08-17:
Incorporated.

PR #53 (2024-08-17): Documentation: M_AXI_*Id omission signals clarification 
Clarification added to doc/olo_axi_master_simple.md and doc/olo_axi_master_full.md

Close #50
commit 64ffe4d2 (2024-08-17): DOC: Clarify omission of *_Id signals for AXI masters

closes #50 
Contributor: @kuriousd
closed 2024-08-17 by obruendl