open-logic/open-logic #64: adding the cdc or rdc checking

neilwang0913, 2024-09-07:
Is there any possibility of adding the clock domain crossing or reset domain crossing ehcking feature?

obruendl, 2024-09-08:
What do you mean exactly?

Running formal verification tools on the cdc components?

The problem with this is that I don't know a free formal tool doing this. If you know one - let me know.

neilwang0913, 2024-09-08:
I am not sure the formal verification, including CDC and RDC checking. As an IP block, do we need to do this basic check? what is the free formal tool you are using?

obruendl, 2024-09-12:
I am not using any formal tool yet. I sm investigating some formal tools right now.

If your question wad not about formal CDC checking, I obviously misunderstood. Could you clarify your initial question?

obruendl, 2024-09-27:
As I did not get any reply, I will proceed to close the issue. Feel free to re-open if it still is active.

closed 2024-09-27 by obruendl