open-logic/open-logic #68: Question: Spurious "bit" in MISO line after first byte (SPI Slave)

betocool-prog, 2024-09-19:
Hey all,

I started using the SPI slave component a few days ago with good results so far. To confirm that things are working I'm looking at the SPI signals running at about 1Mbps on a Digilent Analog 3 Logic Analyzer.

I noticed that after receiving the first byte, I get a half-bit at the end of the MISO line, as shown below:
[image]

The settings are:
```
spi_component : entity work.olo_intf_spi_slave
generic map (
    TransWidth_g                => 8,
    ConsecutiveTransactions_g   => true
)
```

Everything else is default. I'll still look into it in case it's my settings, but perhaps someone has seen this before?

Cheers,

Alberto

betocool-prog, 2024-09-19:
I also just noticed looking at the graph above that MISO and MOSI are shifted. MOSI seems to have the correct CPOL=0, CPHA=0 setting, where as MISO sends the data out on the rising edge. I will set the SpiCPHA_g and SpiCPOL_g values explicitly to zero to see if that changes things.

obruendl, 2024-09-19:
**Shift of MISO/MOSI**

This is intended - see [here](https://github.com/open-logic/open-logic/blob/main/doc/intf/olo_intf_spi_slave.md#spicpol_g-and-spicpha_g). However, I am open to discuss if this is a good idea or if I overlooked something.

Because the slave only asserts MISO a few clock cycle *after* the sampling edge of the clock was detected and this edge had to propagate to the slave first and to be synchonized there, we can be 100% sure that this data does not toggle at the master before the same sampling edge. Hence hold time is always given. So there is no actual need for shifting data application and data sampling by half a clock for the slave. Hence IMO this is "safe and allowed" - if you disagree, please let me know what risks/problems you see.

Why do I think it is a good idea? Well, the more you tune up SCLK frequency, the more the SCLK->MISO propagation delay of 4 clock cycles degrades timing. Transmitting data on the sampling edge gives us half a clock cycle more margin and allows reaching higher SCLK speeds - and hence higher throughput.

Please provide your feedback on this topic.

obruendl, 2024-09-19:
**Half Bit at the end of the transaction**

The *olo_intf_spi_slave* fetches the next data-byte to transmit after a transaction completed. Until the new data is latched, it doesn not know what to transmit through MISO and the behavior is undefined. IMO this is "allowed" (on the sampling edge data is correct). 

However, the needless toggling is somewhat confusing - and also longer than required. The new MISO could be applied immediately after Tx_Valid='1' (currently it is applied only on the SCLK edge).

Again, I think the behavior is valid (i.e. does not lead to any functional problems) but I decided to optimize this behavior - just to avoid confusion. You can try the updated version on the *develop* branch.

obruendl, 2024-09-19:
BTW: As you seem to successfully use Open Logic, it would be great if you do a post about your experience with it.

betocool-prog, 2024-09-19:
Hey, thanks for looking into it! I will have a look into the changes. I'll be away for a couple of weeks from today, but I will get into the development right after.

I'll try to find a meaningful example to write a post, let you know, and you can refer to it if you like it.

Cheers,

Alberto

obruendl, 2024-09-19:
Cool.

To my understanding you just asked because you wondered but you don't see any functional problems. Correct?

If so, would it be fine for you if I proceed to close the issue?

closed 2024-09-20 by obruendl