open-logic/open-logic #75: Minor improvements

tasgomes, 2024-10-08:
- On each block VHDL source code, please consider adding an hyperlink to the documentation of it. Makes it more friendly to navigate from source code to documentation in GitHub. This is similar to what you already do with having a hyperlink in the doc to the source code.
- The folder where FIFO packet TBs are located are missing the word "fifo"
[image]

- Consider using only spaces for tabs and avoid mixing both. Code is harder to read when both are mixed (you can probably set this in your code editor so that only spaces are used)
[image]

[image]

obruendl, 2024-10-14:
@tasgomes 

Points 2 and 3 are fixed - as expected the linter catched a few tabs that sneaked in :-)

In case you are interested in VSG linting and how it can nicely be integrated with VSCode, you might want to have a look on the HowTo document.

Point 1 will follow later.

obruendl, 2024-10-24:
@tasgomes 

Point 1 is applied as well (still in develop ATM). Hence I close the issue. Feel free to reopen if you feel something missing.

closed 2024-10-24 by obruendl