open-logic/open-logic #79: olo_base_fifo_async: Wrong CDC direction for read address

aleschx, 2024-10-16:
https://github.com/open-logic/open-logic/blob/5d3b8c6f0a96f806601850808edf915804becbf7/src/base/vhdl/olo_base_fifo_async.vhd#L275-L287

We think the clocks/resets should be reversed as we get CDC timing errors otherwise.

obruendl, 2024-10-16:
# Bugfix Release Checklist

* [x] Prepare release branch
* [x] Check clean Git History
* [x] Run coverage simulations locally -> Check 100% coverage
  * Partial - Only for the fifo async modified
* SKIP - Test tutorial builds
  * olo_base_fifo_async affected is not part of tutorials
* [x] Update release notes
* [x] Update Fusesoc Corefiles (run tools/fusesoc/UpdateCoreFiles.py --version x.y.z)
* [x] Push (commit message: DOC: update files for release)
* [ ] Merge to main (on GitHub through PR), commit message *RELEASE: x.y.z*
* [ ] Set Tag *x.y.z*
* [ ] Test fusesoc build 
    * Create new fusesoc dir and add open-logic
    * [ ] fusesoc library add open-logic https://github.com/open-logic/open-logic
    * [ ] fusesoc run --tool vivado --target zybo_z7 open-logic:tutorials:vivado_tutorial:2.1.0
    * [ ] fusesoc run --tool quartus --target de0_cv open-logic:tutorials:quartus_tutorial:2.1.0
* [ ] Update badges (includes a coverage simulation) - on the tagged commit
* [ ] Create release (on github, from tag)
* [ ] Update fusesoc library (https://github.com/fusesoc/fusesoc-cores)

obruendl, 2024-10-16:
@aleschx aleschx

I release 2.3.1 bugfix release. I could reproduce the bug and fix it - hence I close the issue.

Please reopen the ticket if the issues on your side persist.

closed 2024-10-16 by obruendl