open-logic/open-logic #81: Provide SV assertion based tests

evlog, 2024-10-21:
I would like to contribute some SV testbenches for some of your blocks, just to master a bit better the SVA concept. Is the use of SV code for testing acceptable to be committed to the repo?

obruendl, 2024-10-24:
I think this boils down to the question whether Verilog Testbenches are added to open-logic. 

While I would see a value of having Verilog Testbenches to check if all entities (to make real sense ALL entities should be covered, having only some looks quite incomplete to me) really can be instantiated from Verilog, I fear the effort of maintaining a second set of testbenches. 

Up to now my idea for checking if all entities really can be instantiated from Verilog (theoretically they should, I also have tutorials but I did not run tests on every component) was to do this through the synthesis tools as soon as I have build slave which can run test-synthesis for all blocks with all supported tools. This build runner is an Idea I have for long but I currently lack funding for buying one.

What I potentially would see as good middle ground: Having a minimal verilog testbench for each block (only absolutely minimal testcase to show it works as intended). Functional testing with full coverage and larger test-cases shall stay in VHDL. WHat do you think about this idea?

If we should decide to do this: Which Simulator would you use? I am currently not aware of any open source simulator allowing mixed language.

evlog, 2024-10-24:
Well I have made some trials with the Vivado simulator and the AXI master, because it also has the integrated wave editor. I am not sure if Verilator can also handle mixed language dut/testbench scenarios,

obruendl, 2024-10-24:
AFAIK verilator is Verilog only.

I already did a small mixed simulation turorial for Questa (see [here](https://github.com/open-logic/open-logic/blob/feature/doc-fixes/doc/tutorials/QuestasimTutorial.md)). This can be run with a free altera questa license.

You can try Vivado simulator. My finding was that it is not very easy to get it running scripted (which will be required to run simulations on all blocks).

obruendl, 2024-11-04:
@evlog 

Is there any update on this issue? Unless there is real progress and we can define a set of changes you would contribute and I see as reasonable for going into the next release, I would close the issue. Currently it looks more like an "idea" - not an "issue". A valid idea but I would like to avoid having open issues for every feature idea - I tend to track those ideas in the wiki instead.

evlog, 2024-11-04:
Please check this repo with a tb that I coded. Let me know your thoughts. Please feel free to close this thread, cause as you said this is not a bug or issue.

https://github.com/pebble-technologies/AXI-master-IP-SV-testbench/tree/main

obruendl, 2024-11-04:
@evlog 

I had a short look at the TB.

Let me ask a few questions:

1. Where do you apply m_axi_awready='1'? To me it looks like this signal is unassigned - and executing the testbench in QuestaSim also looks like this is the case:

[image]

2. To me it looks like you hardcoded the expectation that the axi aw and w channels become valid exactly one clock cycle after cmd_wr_valid='1':

[image]

First of all, the specification of olo_axi_master_full does not enforce this. olo_axi_master_full might have some latency and it would not be a breaking change if this latency changes by a single clock cycle. Hence a TB should not depend on the latency being exactly one clock cycle.

In fact I know it has some latency and if I run the simulation, your TB is throwing some errors although I know the olo_axi_master_full to be working correctly:

[image]

3. Around the lines I questioned in (2), I believe you should check for m_axi_awvalid='1' before looking at any other signals. Otherwise AXI handshaking is not respected and all remaining checks of m_axi_aw_... signals do not actually make sense.

I did not investigate more from here on. 

Let me know if I misunderstood something. I am not a system-verilog expert and VHDL is my more natural language.

obruendl, 2024-11-08:
@evlog 

Any update on this one?

evlog, 2024-11-08:
Sorry for the delay. Please close the issue. I will contact you in person when I have more updates.

closed 2024-11-12 by obruendl