open-logic/open-logic #85: Question: We Implement Reduce Functions

andkae, 2024-10-25:
Hello Together,

this looks like a real cool project, but why implementing ```reduce``` functions in:

https://github.com/open-logic/open-logic/blob/8b025ca8dea5a65fad18100fa751a48e1407b0d0/src/base/vhdl/olo_base_pkg_logic.vhd#L222

There are implemented in ```ieee.std_logic_misc.all```, see here: https://www2.cs.sfu.ca/~ggbaker/reference/std_logic/src/std_logic_misc.vhd

Thanks for your help.

BR,
Andreas

obruendl, 2024-10-25:
Don't ask "why" if you detect nonsense. Just propose a change ;-)

I appreciate suggestionw for improvement.

Reason for things being as they are: 
We are all humans andcwe don't know everything.

Suggested fix
I plan a not 100% backward compatible change in the next release anyways due to the introduction of VSG linting. I will try removing the needless function in the same release. In case of any tool incompatibilities, I would come back to you.

Sane?

obruendl, 2024-10-27:
Implemented. Will be contained in next release.

Hence I close the issue.

closed 2024-10-27 by obruendl