open-logic/open-logic #101: Dangerous asynchronous reset

tasgomes, 2024-11-27:
Isn't this a dangerous approach of synchronizing an asynchronous reset?

https://github.com/open-logic/open-logic/blob/1d24b7928ba63e67d7fea8f2651902311a11addf/src/base/vhdl/olo_base_reset_gen.vhd#L63-L73

Many flip-flops (FFs) use a synchronous reset input, i.e., they will have `RstSync` as their reset synchronous signal. This means `RstSync` is part of the logic that goes to the D-input of those FFs. Hence, this signal should respect the setup and hold requirements of the FFs.

If `RstIn` is asserted, it means `RstSync` will be asynchronously asserted as well (according to the logic implemented), causing the D-input of those FFs to be set asynchronously to the clock. This may result in metastability on their Q-output. Everything connected to the Q-output will experience the metastability, and if those registers (connected to the Q-output) are not reset at all (or belong to another clock/reset domain), then metastability (or incorrect values) can propagate throughout the logic.

Wouldn't it be safer to pass `RstIn` through a 2-FF synchronizer first, and then generate `RstSync` always synchronously? This would be a safer solution regardless of what is connected to the Q-output of those FFs, ensuring the design is fully synchronous.

Another point:

Can you verify if the description matches the implementation?

Description:

[image]

Code:

https://github.com/open-logic/open-logic/blob/1d24b7928ba63e67d7fea8f2651902311a11addf/src/base/vhdl/olo_base_reset_gen.vhd#L109-L111

If `RstPulseCycles_g ` = 3, then the reset out depends on `RstSync`  which from line 73 and 67, we can see it can be set asynchronously.

https://github.com/open-logic/open-logic/blob/1d24b7928ba63e67d7fea8f2651902311a11addf/src/base/vhdl/olo_base_reset_gen.vhd#L73

https://github.com/open-logic/open-logic/blob/1d24b7928ba63e67d7fea8f2651902311a11addf/src/base/vhdl/olo_base_reset_gen.vhd#L63-L67

tasgomes, 2024-11-27:
It is clear that de-assertion being always synchronous is a must. 

I am talking about the assertion being asynchronous, which can cause metastability.

> Conclusion in a Xilinx Forum was the same:
https://adaptivesupport.amd.com/s/question/0D52E00006hpKqxSAE/reset-asynchronous-assert-synchronous-deassert-why-asynchronous-assert?language=en_US

In there, avrumw also confirms that:

> You are right - if your "local" flip-flops have synchronous set/reset, then there is (almost) no advantage in the asynchronous assertion of the S/R - **in fact it is technically a timing violation**

He confirms this is still a timing violation - that may generate metastability.

It is probably safe 99% of the time, if all the subsequent logic is running on the same reset and it is reset one or two clocks later. But I still worry about the other logic that is not running on the same reset domain, or not being reset at all, and is fed by these registers that can generate metastability for one clock cycle. I do see the possibility of the metastability effects being propagated to other parts of the design due to this asynchronous assertion of the reset.

> There are (very rate but existing) cases where an asynchronous assertion is wanted (e.g. resetting I/Os without clock). Also I am open minded regarding users of open-logic. Some of them may want to use asynchronous resets (I have a different opinion, but I don't want to exclude them from my user community just for that opinion).

I agree that there are very rare cases where an asynchronous assertion is wanted but I am afraid your code will be 95% of the time used in cases where this need is nonexistent, potentially exposing this issue.

At the end, it is the designer responsibility to implement proper reset mechanisms depending on the application. 

3 key points about this:

1. Vivado does not warn us about this critical path unless the constraint `config_timing_analysis -enable_preset_clear_arcs true` is added to the project. 
2. If one connects a reset that is asserted asynchronously to the D-input of FF, it is their responsibility to ensure other relevant logic is also held in reset until the reset is de-asserted.
3. This could all be avoided by using a reset that is synchronous assert and synchronous de-assert, as long the logic does not have to be reset if clock is lost.

My goal was to open the discussion to bring awareness to everybody. Even though the risk is very small, I would probably avoid the use of this scenario.

obruendl, 2024-11-27:
I get it now. I don't believe the chance of real-life errors is super-low, but I get the point with the receiving FF not being on the same reset domain.

I implemented a fix. Please review. Please also review the doc.

https://github.com/open-logic/open-logic/commit/16d5cca9fa21f0e531d11b64af205edd04571958

I first was not sure if having one additional output suffices or if a double stage synchronizer is required. However, I concluded it is sufficient because:
 If the reset occurs so close to an edge that the first FF gets metastable, the metastability will be removed by the second FF (which acts as double stage synchronizer).

Let me know if you agree. If not, we can discuss if I add a "olo_intf_sync" at the input. However, I would avoid this if possible to avoid unnecessary delays on the reset - and I would want to keep detection asynchronous so short reset pulses are detected even with slow clocks. Can be discussed as well if really required.

obruendl, 2024-11-28:
@tasgomes 

I think I now got it implemented the right way. I know that we probably do not 100% agree on the "detect asynchronously" topic - but as long as you don't see any safety criticality on it, I think this should not be a roadblock.

For the case _AsyncResetOutput_g=False_ the reset is now properly synchronized (also the assertion) with a multi-stage synchronizer. The synchronizer has all reequired attributes and comes with scoped constraints.

Please check and provide feedback.

https://github.com/open-logic/open-logic/blob/feature/reset-gen-sync/doc/base/olo_base_reset_gen.md

tasgomes, 2024-11-29:
@obruendl I think now this is safer.

closed 2024-11-29 by obruendl