open-logic/open-logic #98: Improving import_sources.tcl

tasgomes, 2024-11-27:
I followed the guide described in [use-open-logic-in-a-amd-vivado-project](https://github.com/open-logic/open-logic/blob/main/doc/HowTo.md#use-open-logic-in-a-amd-vivado-project) to add the `olo` library to my Vivado project. 

This results on all files to be added, even the ones I don't need for my Vivado project:

[image]

Typically, I prefer to only add the minimum files required by the Vivado project. This helps on keeping the Vivado project tidy (e.g. in the picture above I would only see my top-level and the non-used `olo` files would not be visible in this view). 

I do like that all the scoped constraints are added automatically, and this only represents one TCL file in my project:

[image]

Would it be possible to enhance the [import_sources.tcl](https://github.com/open-logic/open-logic/blob/main/tools/vivado/import_sources.tcl) with input arguments so we can keep adding `olo` modules only as necessary? 

Alternatively, could we separate the setting up of the constraints and let the user add the VHDL files manually?

obruendl, 2024-11-27:
Suggestions:

a) The most obvious way of doing is that you just run *import_sources.tcl* and then remove whatever you don't need.

b) What I could add easily, is that you can select which *area* of open logic you want to import (base, intf, axi) - but you would still get all files within the areas selected.

Resolving all dependencies in that TCL script would be effort and maintenance intense. And because it would have to be done for all tools/vendors resp. shared between them. I actually don't want open logic to become a big TCL project - more is less.

Do either a) or b) work for you or do you have any other concrete suggestion?

tasgomes, 2024-11-27:
b) would not solve it, so a) it is.

obruendl, 2024-11-27:
Ok. I'll add this to the documentation and keep the issue open until this is done.

obruendl, 2024-11-27:
Done.

closed 2024-11-27 by obruendl