open-logic/open-logic #99: CDC violation on olo_intf_clk_meas

tasgomes, 2024-11-27:
The `Rst` signal is synchronous to `Clk`. However, in the code below, it is being used on the `ClkTest` clock domain to synchronously reset the counter value (`CntrTest_T` and `ResultValid_T`) without any proper CDC logic or without any timing exception:

https://github.com/open-logic/open-logic/blob/1d24b7928ba63e67d7fea8f2651902311a11addf/src/intf/vhdl/olo_intf_clk_meas.vhd#L122-L146

If the clock frequencies between `Clk` and `ClkTest` are very similar, let's say, 100 MHz vs 99 MHz, this will generate timing issues (as timing cannot be met). Even if timing is met, tools will complain that the clocks are timed together, i.e., the path is not excluded from timing analysis.

Please review.

obruendl, 2024-11-27:
That's a bug of course. 

I added a clock-crossed reset signal `Rst_T` but simply used the wrong one. Bugfix is implemented here #100 . Please review and let me know if you agree with the fix.

obruendl, 2024-12-01:
Fixed in Release 3.0.2 I just published. Hence I close the ticket.

closed 2024-12-01 by obruendl