open-logic/open-logic #107: [FEATURE REQUEST] Asymetric fifo

rbrglez, 2024-12-05:
From what I can observe, all the implemented FIFOs are symmetric, meaning the input and output data sizes are the same. It would be great to have a FIFO implementation where the input and output data sizes are different.

If you agree that this FIFO could be a valuable addition, I'd be happy to implement it if you're short on time.

obruendl, 2024-12-05:
For my understanding, qhat would be the main advantage of an asymmetric FIFO over combining a width convwrter and a symmetric FIFO?

I am very open for contributions. See contributors documentation and the video regarding contributions on the youtube channel. 

If you want to contribute the FIFO, I suggest first writing a draft spec so we can discuss concept and interface before diving inro code. Ok?

rbrglez, 2024-12-06:
Oh... Looks like I should read entity list more thoroughly. I'll combine the width converter with the symmetric FIFO. If the asymmetric FIFO unit is still missing, I'll create a new issue. For now, I'm closing this one.

closed 2024-12-06 by rbrglez

obruendl, 2024-12-06:
Ok.

As long as combining the two things (FIFO + Width Converter) does not miss significant features, this approach seems more flexible to me.

You can choose if you want a sync-, async- or packet-fifo and you can do some resource/throughput tradeoffs by doing the width conversion on one or the other side of the fifo.