open-logic/open-logic #109: [Fusesoc] Redundant provider key in core files

rbrglez, 2024-12-22:
## 1. Summary

When adding the `open-logic` repository to a `fusesoc` project, the entire repository is downloaded, but `fusesoc core list` displays the cores as empty, even though the source code for these cores is already present in the project directory. Additionally, running the project causes `fusesoc` to download the `open-logic `cores again into `~/.cache/fusesoc`, even though they are already available locally. The issue is identified to be caused by the presence of the `provider` key in the `open-logic` `*.core` files

Excerpt from [olo_base.core](https://github.com/open-logic/open-logic/blob/main/src/base/olo_base.core):
```yml
CAPI=2:

name : open-logic:open-logic:base:3.0.2
description : ...

filesets:
  rtl:
    files:
      - vhdl/olo_base_cc_n2xn.vhd
      - ...
      - vhdl/olo_base_cc_xn2n.vhd
    file_type : vhdlSource
    logical_name : olo

targets:
  default:
    filesets :
      - rtl
  
provider:
  name    : github
  user    : open-logic
  repo    : open-logic
  version : 3.0.2
```

The core files in the `open-logic` repository, include a `provider` key. This key points to the GitHub repository containing the source code, which is necessary for example the [fusesoc-cores](https://github.com/fusesoc/fusesoc-cores) repository (where only core files are present, not the source code). However, in the context of the current project, this key should not be included because the source code for the `open-logic` cores is already locally available.

## 2. Example Configuration

Adding `open-logic` to my project
```bash
fusesoc library add open-logic https://github.com/open-logic/open-logic
``` 

### 2.1 Configuration with `provider` key

Listing all my cores in project. Note that `open-logic` cores are listed as empty, even though whole repository was downloaded into my project.
```bash
$ fusesoc core list

Available cores:

Core                                         Cache status  Description
================================================================================
::vga_timing_core:0.1.0                     :      local : VGA Timing Core
::xbc_testbench_utils:0.1.0                 :      local : XBC Testbench Utility Package
open-logic:open-logic:axi:3.0.2             :      empty : AXI related modules see https://github.com/open-logic/open-logic/blob/main/doc/EntityList.md#axi
open-logic:open-logic:base:3.0.2            :      empty : Basic Circuitry (e.g. FIFOs, CDCs, ...) see https://github.com/open-logic/open-logic/blob/main/doc/EntityList.md#base
open-logic:open-logic:intf:3.0.2            :      empty : Interfaces (e.g. I2C, synchronizer, SPI, ...) see https://github.com/open-logic/open-logic/blob/main/doc/EntityList.md#intf
open-logic:tutorials:quartus_tutorial:3.0.2 :      empty : quartus tutorial for open-logic, targetting DE0-CV board
open-logic:tutorials:vivado_tutorial:3.0.2  :      empty : vivado tutorial for open-logic, targetting Zybo Z7-10 board

``` 
If I run my core, which depends on `open-logic:open:logic:base`, you can see that `base` is downloaded again into `~/.cache/fusesoc` even though source code is already present in `~/my_project/fusesoc_libraries/open-logic`
```bash
$ fusesoc run --target project --setup vga_timing_core
INFO: Preparing open-logic:open-logic:base:3.0.2
INFO: Downloading open-logic/open-logic from github
INFO: Preparing ::xbc_testbench_utils:0.1.0
INFO: Preparing ::vga_timing_core:0.1.0
INFO: Setting up project
```
### 2.2 Configuration without `provider` key

If I remove `provider` key from `*.cores`, my project looks like this:
```bash
$ fusesoc core list

Available cores:

Core                                         Cache status  Description
================================================================================
::vga_timing_core:0.1.0                     :      local : VGA Timing Core
::xbc_testbench_utils:0.1.0                 :      local : XBC Testbench Utility Package
open-logic:open-logic:axi:3.0.2             :      local : AXI related modules see https://github.com/open-logic/open-logic/blob/main/doc/EntityList.md#axi
open-logic:open-logic:base:3.0.2            :      local : Basic Circuitry (e.g. FIFOs, CDCs, ...) see https://github.com/open-logic/open-logic/blob/main/doc/EntityList.md#base
open-logic:open-logic:intf:3.0.2            :      local : Interfaces (e.g. I2C, synchronizer, SPI, ...) see https://github.com/open-logic/open-logic/blob/main/doc/EntityList.md#intf
open-logic:tutorials:quartus_tutorial:3.0.2 :      local : quartus tutorial for open-logic, targetting DE0-CV board
open-logic:tutorials:vivado_tutorial:3.0.2  :      local : vivado tutorial for open-logic, targetting Zybo Z7-10 board
```

If I run my project now, `base` core isn't downloaded again into `~/.cache/fusesoc`, instead it uses source code already downloaded into my project
```bash
$ fusesoc run --target project --setup vga_timing_core
INFO: Preparing open-logic:open-logic:base:3.0.2
INFO: Preparing ::xbc_testbench_utils:0.1.0
INFO: Preparing ::vga_timing_core:0.1.0
INFO: Setting up project
```

obruendl, 2024-12-24:
@rbrglez 

Thank you for rising the issue. I am not too familiar with fusesoc but I maintain support for it, hence I am glad if users with more fusesoc mileage help me improving things.

Let me check if I understood everything correctly:
* .core files in fusesoc-cores shall have the provider key
* .core files in the open-logic repo shall NOT have the provider key

I checked the .core files of [pulp platform](https://github.com/pulp-platform/axi_mem_if/blob/master/axi_mem_if.core) and there it is done this way, so this seems common and it would be easy to implement.

I have one concern I hope you can help me with: The .core files contain the version. I update the version in .core files upon every release - hence it would be somewhat unclean, if the local files show the version of the latest release even if the actually checked out commit contains changes (e.g. if you have checked out a not yet released version of the *develop* branch). Is there any common way to solve this?

I would see the following options:
* Set version to 0 for the .core files in the repo (for local usage).
  * This is how pulp platform does
  * I find it somewhat unclean, because 0 in semantic versioning means "something very old"
* Use a different VLNV name for the local .core files
  * open-logic:open-logic-local:3.0.2 in the core-files in the repo
  * open-logc:open-logic:3.0.2 for the files containing the provider key (e.g. present in fusesoc-cores)
  * Due to the different VLNV I could provide both .core files in the repo. The user could then choose between *open-logic:open-logic-local* (which means "whatever is checked out locally") and *open-logic:open-logic* (which means "exactly the version that is officially released)

My feelings would go more towards the second option. **Please let me know whats your take on this** so I can implement it in the most optimal way.

rbrglez, 2024-12-24:
## Option 1

> - Set version to 0 for the .core files in the repo (for local usage).
>     - This is how pulp platform does
>     - I find it somewhat unclean, because 0 in semantic versioning means "something very old"

I am not a fan of this option either

## Option 2

> - Use a different VLNV name for the local .core files
>     - open-logic:open-logic-local:3.0.2 in the core-files in the repo
>     - open-logc:open-logic:3.0.2 for the files containing the provider key [...]

Second option is better but you have changed **Name** in VLNV. I would rather change **Version** in VLNV.
In the development branch (e.g. `main`) I would always append `-dev` to every version number, so that it would look like this: `4.0.0-dev`. When development is finished and you want to release it for example as a version `3.1.0`, you merge development branch `main` to the release branch `release` and change all the `4.0.0-dev` versions to the official version you want to release as  (e.g., `3.1.0`) and then you create a release on github with tag `3.1.0`

### **NOTE**:  cores are NOT accessible via fusesoc provider key until release is created

I encountered an error when I pushed .core files with version `3.1.0` to my fork of the open-logic repository, but did not create a release. When I attempted to fetch the new version using another .core file with the provider key via the fusesoc `fetch` command, I received the following error message.

```bash
$ fusesoc fetch base
INFO: Downloading rbrglez/open-logic from github
ERROR: Failed to fetch 'open-logic:open-logic:base:3.1.0': Failed to download 'https://github.com/rbrglez/open-logic/archive/3.1.0.tar.gz'. 'Not Found'
```

Below is an excerpt from my testing .core file, which attempts to download version 3.1.0 from my forked open-logic repository:

```yaml
CAPI=2:

name : open-logic:open-logic:base:3.1.0
description : ...

filesets:
  rtl:
    files:
      - vhdl/olo_base_cc_n2xn.vhd
      - ...
      - vhdl/olo_base_cc_xn2n.vhd
    file_type : vhdlSource
    logical_name : olo

targets:
  default:
    filesets :
      - rtl
  
provider:
  name    : github
  user    : rbrglez
  repo    : open-logic
  version : 3.1.0
```

Only after creating a release on github was I able to fetch the core with version `3.1.0`

### Example of checking out development commit

All of the development branches should have `-dev` suffix on every version. That way you would immediately see that this isn't official release.

```bash
$ fusesoc core list                                                              
                                                                                                                               
Available cores:                                                                                                               
                                                                                                                               
Core                                             Cache status  Description                                                     
================================================================================                                               
open-logic:open-logic:axi:4.0.0-dev             :      local : AXI related modules see https://github.com/open-logic/open-logic/blob/main/doc/EntityList.md#axi
open-logic:open-logic:base:4.0.0-dev            :      local : Basic Circuitry (e.g. FIFOs, CDCs, ...) see https://github.com/open-logic/open-logic/blob/main/doc/EntityList.md#base                                 
open-logic:open-logic:intf:4.0.0-dev            :      local : Interfaces (e.g. I2C, synchronizer, SPI, ...) see https://github.com/open-logic/open-logic/blob/main/doc/EntityList.md#intf
open-logic:tutorials:quartus_tutorial:4.0.0-dev :      local : quartus tutorial for open-logic, targetting DE0-CV board        
open-logic:tutorials:vivado_tutorial:4.0.0-dev  :      local : vivado tutorial for open-logic, targetting Zybo Z7-10 board    
```

obruendl, 2024-12-24:
Yeah, I am aware that without a release it does not work. I ran into this before - it's a problem for me because I cannot test fusesoc builds BEFORE releasing.

I like your suggestion with the -dev appendix to the release. The reason I did suggest to change names is that I am unsure what letters in the version do to semantic versioning. Open logic axi has a dependency to open logic base. Will fusesoc think that 3.0.2-dev is newer than 3.0 2? I'd like to avoid that fusesoc mixes local files with diwnloaded releases if both core files are present.

Is the behavior of dependencies documented for version numbers including letters?

You get the mood: I like the idea but I would like to fix it once and better ask one more question up-front than changing again later.

rbrglez, 2024-12-25:
> You get the mood: I like the idea but I would like to fix it once and better ask one more question up-front than changing again later.

Yeah, I Agree. Thanks to your questions I found out that versions with `-dev` suffix don't work. If I try to run a target, which depends on a core which has version with `-dev` suffix fusesoc enters an infinite loop.

I have created testing repository where open-logic cores with different versions can be tested: https://github.com/rbrglez/testing-fusesoc-versioning

Here is an excerpt from testing repository README

## 1) official release version 3.2.0 is added last
#### Command

```bash
./init_development_olo_3.2.0.sh
./init_release_olo_3.2.0.sh
./init_test_target_core.sh
fusesoc core list
```

#### Output

```bash
$ fusesoc core list
WARNING: Replacing open-logic:open-logic:intf:3.2.0 in development-cores/open-logic-3.2.0/src/intf with the version found in /home/rbrglez/Documents/local/testing-fusesoc-versioning/release-cores/open-logic-3.2.0
WARNING: Replacing open-logic:open-logic:axi:3.2.0 in development-cores/open-logic-3.2.0/src/axi with the version found in /home/rbrglez/Documents/local/testing-fusesoc-versioning/release-cores/open-logic-3.2.0
WARNING: Replacing open-logic:open-logic:base:3.2.0 in development-cores/open-logic-3.2.0/src/base with the version found in /home/rbrglez/Documents/local/testing-fusesoc-versioning/release-cores/open-logic-3.2.0

Available cores:

Core                                         Cache status  Description
================================================================================
::test-target:0.1.0                         :      local : Dummy target
open-logic:open-logic:axi:3.2.0             :      empty : official release; axi core; 3.2.0
open-logic:open-logic:base:3.2.0            :      empty : official release; base core; 3.2.0
open-logic:open-logic:intf:3.2.0            :      empty : official release; intf core; 3.2.0
open-logic:tutorials:quartus_tutorial:3.2.0 :      local : quartus tutorial for open-logic, targetting DE0-CV board
open-logic:tutorials:vivado_tutorial:3.2.0  :      local : vivado tutorial for open-logic, targetting Zybo Z7-10 board
```
## 2) development version 3.2.0 is added last

#### Command

```bash
./init_release_olo_3.2.0.sh
./init_development_olo_3.2.0.sh
./init_test_target_core.sh
fusesoc core list
```

#### Output

```bash
$ fusesoc core list
WARNING: Replacing open-logic:open-logic:base:3.2.0 in /home/rbrglez/Documents/local/testing-fusesoc-versioning/release-cores/open-logic-3.2.0 with the version found in development-cores/open-logic-3.2.0/src/base
WARNING: Replacing open-logic:open-logic:axi:3.2.0 in /home/rbrglez/Documents/local/testing-fusesoc-versioning/release-cores/open-logic-3.2.0 with the version found in development-cores/open-logic-3.2.0/src/axi
WARNING: Replacing open-logic:open-logic:intf:3.2.0 in /home/rbrglez/Documents/local/testing-fusesoc-versioning/release-cores/open-logic-3.2.0 with the version found in development-cores/open-logic-3.2.0/src/intf

Available cores:

Core                                         Cache status  Description
================================================================================
::test-target:0.1.0                         :      local : Dummy target
open-logic:open-logic:axi:3.2.0             :      local : dummy axi description 3.2.0. This is development version!
open-logic:open-logic:base:3.2.0            :      local : dummy base description 3.2.0. This is development version!
open-logic:open-logic:intf:3.2.0            :      local : dummy intf description 3.2.0. This is development version!
open-logic:tutorials:quartus_tutorial:3.2.0 :      local : quartus tutorial for open-logic, targetting DE0-CV board
open-logic:tutorials:vivado_tutorial:3.2.0  :      local : vivado tutorial for open-logic, targetting Zybo Z7-10 board

```

# My suggestion about versioning

I would suggest that development branch (e.g., `main`) is always one MAJOR or MINOR version ahead from the official release. I would also add a description in cores, that would tell you this is development version.

obruendl, 2024-12-25:
I agree on adding the descriptions - will be implemented.

Making "develop" always one number ahead seems suboptimal to me for the following reasons:

- I don't know ahead if the next release is non-reverse compatible (major increment) has new features (minor increment) or obly is a bugfix (bugfix increment). Hence this is somewhat difficult to maintain. 
-  If I increment prior to release, there still would be unclarity if 3.2.0 means "release 3.2.0" or "some WIP state after 3.1.0"

I personally would see using a different library in the vlnv as more clear. I agree with the "dev" naming, so I would opt for open-logic:open-logic-dev:..

Whenever using the "dev" library this means "release plus ongoing development work".

I would provide both core files, the ones pointing to releases (and always downloading them) and the "dev" ones, using whatever is checked out.

Let me know if you see serious problems with this approach - otherwise I would implement the change this way.

BTW: please report the infinite loop to Olof Kindgren. I could consider an error as acceptable but an infinite loop definetly is a bug that should be fixed.

rbrglez, 2024-12-26:
> I personally would see using a different library in the vlnv as more clear. I agree with the "dev" naming, so I would opt for open-logic:open-logic-dev:..

Yes, this looks like the best solution

> I would provide both core files, the ones pointing to releases (and always downloading them) and the "dev" ones, using whatever is checked out.

How will you implement this? I would prefer if release branch (e.g., `main`) would contain only released cores (cores without `-dev` appendix), while development branches (e.g., `develop`) would contain only development cores (cores with `-dev` appendix). What do you think? 

> BTW: please report the infinite loop to Olof Kindgren. I could consider an error as acceptable but an infinite loop definetly is a bug that should be fixed.

Yes, I will create an issue :)

BTW: open-logic cores can't be used via fusesoc-cores repository. Check out this issue: https://github.com/fusesoc/fusesoc-cores/issues/24

obruendl, 2024-12-27:
I created a branch [fusesoc-dev](https://github.com/open-logic/open-logic/tree/feature/fusesoc-dev) with the proposed solution. Please review and let me know if this works for you.

I tried it on my end and it looks fine. When building the vivado tutorial from the open-logic-dev library, local files are used. when building the same tutorial from the open-logic (release) library cores are downloaded.

I thought about naming the release library _open-logic-stable_ but rejected the idea for backward compatibility reasons (users should be able to get the same functionality (i.e. downloading releases from github) by referring to the same existing library name).

obruendl, 2025-01-01:
Implemented. Hence I close the issue.

closed 2025-01-01 by obruendl