open-logic/open-logic #119: Where is the Lattice Tutorial

RichFPGA, 2025-02-13:
It looks like you don't have a Lattice Tutorial. I'd like to make one How do I contribute?  You can reach back to me at my Work E-mail Rich.Staubin@Arrow.com. hope to hear from you soon.

obruendl, 2025-02-14:
<< e-mail Oliver > Rich, 2024-02-14 >>

Hi Rich,

Thank you for reaching out to me!

Let me first write why there currently is no lattice tutorial and in a second part I will describe my ideas how you can contribute one (which I would appreciate).

**Part 1 - Why there is no lattice tutorial**

For Lattice, the first questions is "for which device family/tool?". They have different tools for different device families. Writing the tutorials takes time, so having to write 4 of them for one vendor (Propel/Diamond/Radiant/ICEcube) wasn't exactly boosting my motivation.

I got an iceBreaker Kit donated by 3rd party (not Lattice itself) and I tried building a ICE40 tutorial (ICEcube). But because ICEcube does not come with an integrated programmer and I had to use the open-source one, which somehow bricked the HW. Which again was quite a bummer.

Result was that overall my Motivation for Lattice was a bit low to be honest and since up to now the community did not ask actively for it, I preferred to invest my time into other things. The project is pure hobby and not funded yet - hence I take myself the freedom to work on things that are either urgent (bugs are fixed, whether it is fun or not) or that I like to do.

Note that this does NOT mean that I don't want Lattice tutorials. I am very interested in having some.

**Part 2 - How to  get Lattice tutorials going**

Since you ask from your job e-mail, besides the option that you contribute I'd like to also open the option that Arrow (or Lattice, I don't know your connection) simply pay the work. If it is paid hours I earn money that I can invest back into Open Logic I'd happily write those tutorials - because it would allow me to grow the project.

For contributing:

1. Please let me know what Lattice Device / Tool the tutorial would be for.
2. Please let me know what DevBoard you'd use (should be commercially available to anyone to follow the tutorial). 
3. I'd need one of those boards (donated, since I am not funded yet). I want to be able to reproduce results on my side.
4. There is a video on the "Open Logic" youtube channel about how to contribute and there also is a "Contributors" guide on GitHub. 
5. General flow would be: You fork Open Logic to your GitHub, you create a feature branch (branch off from "develop"). Once you are done you create a Pull Request back to "develop" of the main Repo. I will review and integrate your work once all comments are closed.
6. Besides the Tutorial itself, there are two more things required (7 and 8)
7. Synthesis attributes must be checked to work for lattice (or added). AFAIK Lattice uses SynplifyPRO for synthesis and attributes for SimplifyPRO exist. So I believe this should be fine and only should be double-checked.
8. Tool integration must be added (Open Logic provides a script for each tool that automatically imports all sources into a project). Most tools are TCL scripting based and if the Lattice Tools are as well, this is mostly "copy/paste" from other tool integration scripts plus changing the tool specific TCL commands. Overall not a big effort.

I am looking forward to your response.

Best Regards,

Oliver

obruendl, 2025-03-06:
@RichFPGA 

Do you plan to contribute a lattice tutorial in the coming weeks?

If not, I would like to close the issue and we can re-open it when you decide to contribute a tutorial at some point.

obruendl, 2025-03-11:
@RichFPGA 

Since I did not get a response for a while I close this issue. Feel free to re-open it once you start working on the tutorial.

closed 2025-03-11 by obruendl