open-logic/open-logic #122: RAM initial content

FiberQP, 2025-02-19:
I have a use-case for RAM implementations with initial content. My understanding is that as of right now this is **not** supported.

Im using AMD, reading through UG901 this should be possible even for inferred RAMs.

Is this something you can consider/look into?

obruendl, 2025-02-23:
I will check how to implement it in a way that works for all technologies allowining initialization of ram content.

I am optimistic to get this done.

You can expect this to happen only in 2 weeks. I am busy with carnival (my hobby) next week.

You would of course also be welcome to contribute ram initialization yourself.

obruendl, 2025-03-06:
Back from carnival :-)

Would a generic to pass the initial content as string formatted hex array work for you?

Example below:
...

InitContent_g => "(0xABCD, 0x1234, 0xDEADBEEF)",
...

obruendl, 2025-03-06:
Exactly. Assigning a constant would be the usecase I see most likely. You can easily generate a package containing such an array constant from some data source.

Btw: I am working on fixed point math inclzding python models of all fixed-point open logic conponents. In that context I guess I will imlpement a code generator for generating packages from python.

FiberQP, 2025-03-06:
Cool!
Are you using fix-point library or rolling your own?
In any case, checkout apytypes https://github.com/apytypes/apytypes

obruendl, 2025-03-06:
I'll use en_cl_fix (you find it on github). It contains the same basic functions in VHDL and Python. No need to reinvent this wheel... I'll rather just use the existing wheel and build my car around it (more verilog support and more advanced functions like filters etc.)

obruendl, 2025-03-10:
@FiberQP 

I have implemented RAM initialization on the branch [feature/ram-init](https://github.com/open-logic/open-logic/tree/feature/ram-init). Have a look.

I decided to go with an initializer string without brackets (e.g. "0xABCD, 0x1234, 0xBEEF").

Note: I coded it today and simulation works in GHDL. But I didn't check synthesis results yet or run simulations in Modelsim (I have to renew modelsim license first).

Could you check if the solution implemented works for you and synthesis results at least on AMD are as expected?

FiberQP, 2025-03-11:
Cool, I will try to check it out in the coming days.

obruendl, 2025-03-15:
I tested it for AMD (Vivado) - and everything works as expected.

However, on Quartus RAM initialization does not work. I asked for help in the forum:
https://community.intel.com/t5/Intel-Quartus-Prime-Software/RAM-inference-not-working-for-initialized-RAM/m-p/1675190#M85805

obruendl, 2025-03-15:
@tasgomes You are pretty black-belted with FPGA design tools. Do you wee why Quartus does fail to infer BRAM as soon as I add an initialization value (see https://community.intel.com/t5/Intel-Quartus-Prime-Software/RAM-inference-not-working-for-initialized-RAM/m-p/1675190#M85805)?

obruendl, 2025-03-17:
Updated test results:

- Vivado (AMD): works
- Quartus (Altera): does not work
- Libero (Microchip): works
- Efinity (Efinix): works
- Gowin EDA (Gowin): works

tasgomes, 2025-03-17:
@obruendl I don't see a reason for that. I tried it myself and Quartus inferred a BRAM on my test project:

[image]

What version of Quartus are you using? How did you set the Depth and Width generics? I used `Depth_g = 256` and `Width_g = 8`.

tasgomes, 2025-03-17:
Wait a moment, I just realized I forgot to init the ram contents which is the whole point 🤼‍♂

tasgomes, 2025-03-17:
I tried your setup posted on the forum:

[image]

And M9Ks are inferred.

obruendl, 2025-03-17:
@tasgomes Yes, M9Ks are inferred but only for part of the memory. At least on my side I get some M9Ks + thousands of FFs. In your screenshot I cannot see if you got FFs as well - but I can see the number of "Memory Bits" being 1792 although for 256x16 4k would be required.

Coudl you check if you also see the thousands of FFs besieds the M9Ks?

tasgomes, 2025-03-17:
For your code the `shared variable vs signal` has an effect only when `RamBehavior_g = "WBR"`. 
When it is set to `RBW`, then being `shared variable` or `signal` plays no role.

tasgomes, 2025-03-17:
And oddly, unrolling the for-loop also solves the problem with the byte enable logic:

```vhdl
                    if Be(0) = '1' then
                        Mem_v(to_integer(unsigned(Addr)))(7 downto 0) <= WrData(7 downto 0);
                    end if;

                    if Be(1) = '1' then
                        Mem_v(to_integer(unsigned(Addr)))(15 downto 8) <= WrData(15 downto 8);
                    end if;

                    -- Loop over all bytes
                    --for byte in 0 to BeCount_c - 1 loop
                    --    if Be(byte) = '1' then
                    --        Mem_v(to_integer(unsigned(Addr)))(byte * 8 + 7 downto byte * 8) <= WrData(byte * 8 + 7 downto byte * 8);
                    --    end if;
                    --end loop;
```

Now, the `olo_base_ram_sp.vhd` correctly infers a RAM in Quartus.

obruendl, 2025-03-17:
@tasgomes 

Thank you for investigating. I can confirm it is only the byte-enable logic breaking inference. 

This is especially confusing because `UseByteEnable_g` is set to false (so the problematic code is not even enabled) and beyond that `Wr_Be` is set to "all ones" (default value) so even if the code was enabled it would write the whole memory...

I don't see any sane way to avoid it. I tried reformulating it a bit but nothing helped. Let's see what Altera says to it.

tasgomes, 2025-03-17:
@obruendl 

>  I can confirm it is only the byte-enable logic breaking inference.

But note this as well

> For your code the shared variable vs signal has an effect only when RamBehavior_g = "WBR".
When it is set to "RBW", then being shared variable or signal plays no role.

So shared variable also plays a role!

tasgomes, 2025-03-17:
@obruendl 

Check this example below, you might need to create a 2D array with a byte structure (granularity) in order for the byte enable to be correctly inferred.

```vhdl
    --  build up 2D array to hold the memory
    type word_t is array (0 to NUM_BYTES-1) of std_logic_vector(BYTE_WIDTH-1 downto 0);
    type ram_t is array (0 to DEPTH-1) of word_t;
````
Source:
https://www.intel.com/content/www/us/en/docs/programmable/683082/24-2/ram-with-byte-enable-signals.html

obruendl, 2025-03-17:
@tasgomes 

The shared variable seems fine to me. that RAM cannot be instantiated with "WBR" if the technology implements a "RBW" BlockRAM is to be expected.

I found one solution that works for sure: Implement a RAM without BE and for the case with BE use one of those RAMs per byte
https://github.com/open-logic/open-logic/pull/131

However, this is a bit awkward and may not be 100% efficient (probably for small RAMs it infers one BRAM per byte). Hence I will now try out your suggestion

obruendl, 2025-03-17:
@tasgomes 

I tried the 3D array structure (2D if you go by vectors) you suggested:
https://github.com/open-logic/open-logic/blob/feature/ram-init-3d/src/base/vhdl/olo_base_ram_sdp.vhd

Bad news. This does not solve the issue.

[image]

obruendl, 2025-03-17:
I conclude that this issue is NOT related to RAM initialization. Instead RAM inference with Byte-Enables is broken in Quartus. This was always the case - also before I added RAM initialization. Just that this was only detected now.

@tasgomes Please let me know if you see it differently. You investigated the issue (thanks for that), so I think you are the best sparring partner for this question. 

If you agree with me, I'd proceed to merge the RAM initialization PR - because it does not break anything that was not broken previously. I'd open a new issue instead regarding "RAM inference with BE signals not working for Altera" and document that this feature is not usable due to a Quartus bug.

obruendl, 2025-03-18:
This one is really ugly: Altera just confirmed that RAM inference in Quartus works differently for Quartus Prime _Standard_ and Quartus Prime _Pro_.

https://community.intel.com/t5/Intel-Quartus-Prime-Software/RAM-inference-not-working-for-initialized-RAM/m-p/1675190#M85805

I don't know what other language features and inference patterns differ between the versions. But this is a real bummer to users...

I just detected that there is an issue with TDP RAM inference as well - but without a paid license I cannot even confirm that it will work with the paid version.

FiberQP, 2025-03-20:
Wow, thats kind of unbelievable. Way to go Intel/Altera, shooting your own user base in the foot.

obruendl, 2025-03-21:
@FiberQP 

Well, this only shows how much useful the idea of Open Logic is to provide vendor independent code :-) 

"Challenge Accepted"

The good is not always the easy... FYI: I am now building a script set to run Open Logic components (especially the 3 RAM components) through all supported synthesis tools with different generic settings. This will be a great tool to prove inference - and fix the issue with Altera. 

The infrastructure is up, Altera and AMD tools work. I now add Efinix, Gowin and Microchip - then I go hunting down the Altera inference issues.a

I am on it - but this topic will take a bit of time.

obruendl, 2025-03-26:
@FiberQP / @tasgomes 

I could resolve the issue - thanks to the investigation the issue triggered, I also found a bug that True-dualport RAM (olo_base_ram_tdp) was not synthesizable in Quartus and Gowin. I fixed that one as well.

So now RAM inference is confirmed to work with all tools and RAMs come with the option for initial values (what was the original request).

There is one pitty: The language constructs for RAM inference with byte enables that are understood by all tools are not 100% efficient. I documented this cleanly - but for Open Logic the issue is a minor one because internally in Open Logic byte-enables are not used anywhere.

I am now going to merge this feature to develop - and then probably I am doing a release to fix the inference issues I found on mainline.

obruendl, 2025-03-26:
Feature implemented and merged to develop, hence I close the issue.

closed 2025-03-26 by obruendl