open-logic/open-logic #124: Disable TMR insertion for CDC

svancau, 2025-03-11:
Hello,

I'm working on a FPGA that will require Triple modular redundancy inserted during synthesis.

I'm planning to add the _syn_radhardlevel_ attribute to none, in olo_base_cc_bits and olo_intf_sync.
Other CDC blocks are based on olo_base_cc_bits and additional logic is only in a single clock domain therefore no attribute addition is required. Is this assumption correct ?

For now, only Actel/Microchip/Microsemi FPGAs are supported.

Pull request will follow.

obruendl, 2025-03-11:
olo_base_cc_reset and olo_base_reset_gen also contain synchronizer chains. You should vcover those with the related attribute as well.

I am not too familiar with rad-hardening. For sure TMR does not make sense for synchronizer chains. How about the "syn_radhardlevel=CC" setting? Would this fit for synchronizer chains or not? What is the CC setting even doing - I understand what TMR does but I didn't really understand what the "CC" setting does.

obruendl, 2025-03-12:
As already discussed on LinkedIn:

The changes suggested are non-intrusive. But to stay consistent TMR synthesis capabilities would have to be checked for all supported vendors - which is too much effort given the fact that TMR synthesis is only of interest for a small part of the user-base.

HEnce, I suggest you add these constraints externally (in your constraints file) or you add the attribute in a fork of Open Logic.

closed 2025-03-12 by obruendl

svancau, 2025-03-12:
FYI, the only synthesis attribute is syn_radhardlevel. It exists only with Synopsys Synplify, no other vendors :)