open-logic/open-logic #127: Have blocks with async reset

svancau, 2025-03-11:
Hi,

Some technologies only have async reset inputs and in some cases infer additional logic for resets.
Would it be possible to have a generic that behaves as usual but also allows for async reset ?
Eg, use a procedure to have the same resets with synchronous and asynchronous resets.

@obruendl what do you think of that ?

obruendl, 2025-03-11:
I use synchronous resets only for good reasons - see [this post](https://www.linkedin.com/posts/oliver-br%C3%BCndler-214b69a3_%F0%9D%97%AA%F0%9D%97%B5%F0%9D%98%86-%F0%9D%97%AE%F0%9D%97%BF%F0%9D%97%B2-%F0%9D%97%AE%F0%9D%98%80%F0%9D%98%86%F0%9D%97%BB%F0%9D%97%B0%F0%9D%97%B5%F0%9D%97%BF%F0%9D%97%BC%F0%9D%97%BB%F0%9D%97%BC%F0%9D%98%82%F0%9D%98%80-%F0%9D%97%A5-activity-7233850698566676481-ZSQw?utm_source=share&utm_medium=member_desktop&rcm=ACoAABYQF_ABhBWKN1k5a5GYiISexJOHm0ezc1U). Especially within library components it can't be known what users build around those components and hence I see asynchronous resets as problematic. Chances that I will add asynchronous reset are fairly low - however, I don't want to give you a clear "no" before I fully understood the situation.

I didn't come accross any technology not supporting synchronous resets in my career - and hence my conclusion up to now was that they are virtually for free because supported by virtually every technology.

Which technology are you targetting - which I guess does only support asynchronous resets?

obruendl, 2025-03-12:
As discussed on Linked-In, the proposed solution is that you create a fork and you make the reset asynchronous for the synchronizer chains in there. For all other logic synchronous resets should be OK - they will be built into the LUT logic which seems OK (with the exception of synchronizer chains).

Reason for not adding asynchronous resets: They are dangerous and for future technologies not needed. PROASIC3 you use is pretty oldish - taking big drawbacks for supporting a 20 years old technology is not good tradeoff for the project.

closed 2025-03-12 by obruendl