open-logic/open-logic #129: PWM entity

rbrglez, 2025-03-15:
@obruendl 

I can implement a PWM entity for Open-Logic, as proposed under the [feature ideas](https://github.com/open-logic/open-logic/wiki/Feature-Ideas)

> PWM (symmetric/asymmetric, supporting synchronous current measurement)

However, I’m not entirely sure what is meant by "synchronous current measurement." Could you provide more details or clarify this requirement?

Proposed interface:

```vhdl
entity dev_base_pwm is
    generic(
        PeriodWidth_g : positive;
        DutyWidth_g   : positive
    );
    port(
        Clk : in std_logic;
        Rst : in std_logic;

        In_En     : in std_logic; 
        In_Period : in std_logic_vector(PeriodWidth_g-1 downto 0); -- Period of PWM signal in ticks
        In_Duty   : in std_logic_vector(DutyWidth_g-1 downto 0); -- If Duty = 2**(DutyWidth_g-1), then PWM signal is symmetric

        Out_PeriodFinished : out std_logic; -- Strobe signal indicating when the PWM period has completed
        Out_PeriodCnt      : out std_logic_vector(PeriodWidth_g-1 downto 0); -- Counts from 0 to In_Period - 1
        Out_Pwm            : out std_logic
    );
end entity;
```
Edge cases
 - If `In_Duty = onesVector(DutyWidth_g)`, then `Out_Pwm` is always high
 - If `In_Duty = zerosVector(DutyWidth_g)`, then `Out_Pwm` is always low
 - If `In_Period = zerosVector(PeriodWidth_g)`, then `Out_Pwm` is always low

What do you think about proposed interface? Should I add anything else?

obruendl, 2025-03-16:
I'd suggest to change the generic from _PeriodWidth_g_ to _MaxPeriod_g_. 

  _DutyWidth_g_ can be omitted IMO - Making _In_Duty_ and _In_Period_ the same width would seem lofical to me (to allow 0% ... 100% duty cycle).

The width of _In_Period_ and _In_Duty_ can be calculated based on _log2Ceil()_ from the _olo_base_pkg_math_. 

I'd suggest to use _MaxPeriod_g_ as default value for _In_Period_. I guess in many cases the Period will be constant - hence it would be nice if the _In_Period_ port can be omitted.

Let me know what you think about the suggestions.

rbrglez, 2025-03-17:
Below is the updated interface, with your suggestions implemented:

```vhdl
entity olo_intf_pwm is
    generic(
        MaxPeriod_g : positive
    );
    port(
        Clk : in std_logic;
        Rst : in std_logic;

        In_En     : in std_logic := '1'; 
        In_Period : in std_logic_vector(log2Ceil(MaxPeriod_g)-1 downto 0) := std_logic_vector(to_unsigned(MaxPeriod_g));
        In_Duty   : in std_logic_vector(log2Ceil(MaxPeriod_g)-1 downto 0) := std_logic_vector(to_unsigned(MaxPeriod_g/2));

        Out_PeriodFinished : out std_logic; -- Strobe signal indicating when the PWM period has completed
        Out_PeriodCnt      : out std_logic_vector(log2Ceil(MaxPeriod_g)-1 downto 0); -- Counts from 0 to In_Period - 1
        Out_Pwm            : out std_logic
    );
end entity;
```
I have also set an initial value for the `In_Duty` signal as MaxPeriod_g/2. This ensures that by default, if the `In_Period`, `In_Duty` and `In_En` signals are ommited, `Out_Pwm` will output a PWM signal with a period of MaxPeriod_g and a 50% duty cycle.

Bellow is minimal viable instantiation of PWM unit:

```vhdl
i_pwm : entity work.olo_intf_pwm
    generic map (
        MaxPeriod_g => MaxPeriod_g
    )
    port map (
        Clk     => Clk,
        Rst     => Rst,
        Out_Pwm => Out_Pwm
    );
```

What do you think? Do you have any other suggestions? If not I'll start implementing the unit :D

obruendl, 2025-03-19:
Looks very good.

I'd remove the default value on _In_Duty_ - here the assumption of 50% is as wrong as any other value. I ssuggest to only add defaulr values where there is a default that often is right. E.g. for RAMs I do not define defaults for width/height generics and for the write data port - these all do not have a "natural default".

Probably _Out_PeriodFinished_ could be shortened to _Out_PeriodEnd_ (or start). 

However, those are only "minor comments". From my side you are good to go with implementing it.

rbrglez, 2025-03-20:
I have started implementing PWM unit and I have to decide when should I apply new values of In_Period and In_Duty?

- Option A
Immediately, when the values change. It doesn't matter in which "phase" of period the unit is currently

- Option B
Only at the end of the period.

@obruendl , which option should I choose?

rbrglez, 2025-03-23:
I have implemented PWM unit according to Option A.

I still need to complete the documentation for the PWM unit and add automatic testbench. Currently, I have verified the PWM unit with a manual testbench, which I won't commit.

@obruendl, could you review the pull request: https://github.com/open-logic/open-logic/pull/132

obruendl, 2025-03-31:
@rbrglez 

Regarding application of new In_Period and In_Duty: I'd suggest to latch these values at the beginning of the period.

Reasons: Applying them immediately opens many ugly corner conditions. For example you could produce a much higher frequency than the target PWM frequency when operating In_Duty in certain ways.

Additionally it would be quite some documentation effort to cover all corner cases that immediate application of the values would produce.

It is simply easiest to understand and to completely test the PWM if values are only latched once per period. If the user really cannot wait for updating those values, he can still pull In_Ena low for a clock cycle to immediately restart the period

svancau, 2025-04-03:
Some things I would find awesome for power electronics are:

- Separated timer/reference and comparator, this way we can have any amount of synchronized outputs we need.
- Complementary outputs with dead time
- Multibit PWM output (to drive DDR/Serdes) for increased accuracy :)

I also had some prototype but it still needs more work.

obruendl, 2025-04-07:
@svancau 

I think we have to be clear here what this PWM will be and what it won't be.

IMO this PWM should cover the simple cases like dimming an LED or the like but I would not assume this PWM to be used in power-electronics cases. 

I did a bit of motor control in my career and I see many very specific topics here like:
- Different switching patterns
- Symmetric vs. asymmertric PWM
- Emergency shut-down
- Dead-time, Multi-bit output (as you mentioned)
- Multi-phase handling
- ...

I think adding all of them would overload the entity and still in many cases it finally won't fit. Hence I'd prefer keeping the entity simple (to really be easy to use in simple cases) but documenting that it is not meant as "the golden solution for power electronics". IMO the people building power electronics often have very specific (and application specific) needs for good reasons - and they best gonna implement those themselves. Trying to "rule them all" will not work out according to my experience.

Regarding the "Multiple comparators/outputs": That on I think is a good thing in general. I'd want to get this but I see an interface problem: How to provide multiple _On_Time_ inputs in a way that is working for Verilog and VHDL-93? An array of unconstrained vectors is not an option here. **Do you have a suggestion for this?**

svancau, 2025-05-07:
Hi,

I have no suggestion here, except maybe have a common reference counter and one comparator instance per output channel. This needs a lot of instantiation for such a simple task :(.

obruendl, 2025-06-12:
a PR exists, so I close the issue. The discussions goes into the PR from here.

closed 2025-06-12 by obruendl