open-logic/open-logic #130: List of Typos to Fix in Documentation and Comments

rbrglez, 2025-03-16:
I’m opening this issue to dump all the typos I find in the documentation and code comments. The goal is to collect and track all typos in one place for easier management.

rbrglez, 2025-03-16:
- [ ] The example below shows a ~read~ **write** transaction.
https://github.com/open-logic/open-logic/blob/main/doc/axi/olo_axi_master_full.md#write-transaction

rbrglez, 2025-03-18:
## Document
- 
- https://github.com/open-logic/open-logic/blob/539edf94c10e394ac38db14eb6afc70f5f4e897f/doc/axi/olo_axi_master_simple.md?plain=1#L32

## Typo

since ~he~**the** user has to handle

rbrglez, 2025-03-18:
Typo in directory name `test/base/olo_bae_arb_rr`
https://github.com/open-logic/open-logic/tree/main/test/base/olo_bae_arb_rr

rbrglez, 2025-03-18:
# Document [olo_intf_sync.md](https://github.com/open-logic/open-logic/blob/main/doc/intf/olo_intf_sync.md)

https://github.com/open-logic/open-logic/blob/539edf94c10e394ac38db14eb6afc70f5f4e897f/doc/intf/olo_intf_sync.md?plain=1#L25

- clock crossings in ~_bas_~ **_base_** shall

obruendl, 2025-03-27:
@rbrglez Fixes appliled.

I appreciate documentation reviews and fixes. It is too easy to do typos and often they are difficult to spot for the author himself (at least I am facing this issue).

In future ideally you open a PR with the fixes. Documentation is relateively easy to fix for users.

closed 2025-03-27 by obruendl