open-logic/open-logic #143: olo_intf_spi_master: unsupported function call log2ceil

Donalf, 2025-05-23:
I imported the Open Logic library into an already-existing Vivado 2022.2 project and am trying to use the SPI master code in a block design. However, attempting to add an RTL module block fails with the following error messages. 

The Open Logic module has not been altered since it was imported.

` [IP_Flow 19-627] Port 'Cmd_Slave': XPath expression failed: Unsupported function call or array usage "log2ceil" found in expression "(log2ceil(spirit:decode(id('MODELPARAM_VALUE.SlaveCnt_g'))) - 1)".`

` [IP_Flow 19-627] Port 'Cmd_TransWidth': XPath expression failed: Unsupported function call or array usage "log2ceil" found in expression "(log2ceil((spirit:decode(id('MODELPARAM_VALUE.MaxTransWidth_g')) + 1)) - 1)".`

` [IP_Flow 19-627] Port 'Cmd_TransWidth': XPath expression failed: Unsupported function call or array usage "log2ceil" found in expression "toUslv(spirit:decode(id('MODELPARAM_VALUE.MaxTransWidth_g')), log2ceil((spirit:decode(id('MODELPARAM_VALUE.MaxTransWidth_g')) + 1)))".`

` [IP_Flow 19-4668] Failed to infer definition from module 'olo_intf_spi_master'.`

` [BD 41-1699] Unable to add reference type cell for reference-module 'olo_intf_spi_master'`

` [BD 5-7] Error: running create_bd_cell  -type module -reference olo_intf_spi_master -name olo_intf_spi_master_0 .`

I tested replacing in the VHDL code just this function log2ceil(...) with ceil(log2(...)) and linting shows the following errors:

`Error: type error near slavecnt_g ; current type positive; expected type real`

`Error: found '0' definitions of operator "-", cannot determine exact overloaded matching definition for "-"`

obruendl, 2025-05-23:
Vivado has quite severe restrictions on what calculations on port widths are required for using HDL in BDs. Open Logic does fully support instantiation from HDL but direct integration into graphical tools of the vendors with their restrictions is not off7cially supported.

I suggest you write a wrapper the olo entity that does have fixed port widths and instantiates the Open Logic entity with the desired configuration.

Alternatively, you csn create a custom IP using vivados IP packager where you xan add more elaborate port width calculations in TCL.

patrick-studer, 2025-05-23:
I can feel your pain. Using HDL code directly in Block Designs will create a dummy wrapper around the module code, as it would be packaged properly as IPI-Core.

But for that to work, the top-level HDL file has to follow certain rules - especially no functions to calculate interface widths (like slv size)...
This is why you get XPATH errors.

It is easy to manually package an IPI core with Vivado (in GUI mode).
But the better way is to do this script based, to have it automated.

Maybe you have a look at my IP-Packager Tool for Xilinx.

https://github.com/patrick-studer/xtools/tree/main/ip_packager

Since OLO has grown alot, and not only provides simple HDL blocks to use them inside your code but as standalone in Block Designs, I guess it is time I start to create Xilinx IPI Wrappers for some useful Cores!

Let me know if you have questions abaut IPI-Packaging.

obruendl, 2025-05-24:
@Donalf I fully ageecwith @patrick-studer here: You best use his IP Packager for packaging vivado IP.

@patrick-studer I am open to add a repo with packaged Vicado IP to the organization but I have focus on the runner currently. Would you want to tackle this?

Donalf, 2025-05-28:
> Is there any update on this one or can I close it?
Feel free to close this. Thanks!