open-logic/open-logic #153: new module development: pulse shaper

MiloradPetrovic1, 2025-05-28:
Greetings Oliver!

Regarding the 'Pulse Shaper' entity from the [Feature Ideas list](https://github.com/open-logic/open-logic/wiki/Feature-Ideas), can you provide some detail regarding what you had in mind?

My assumption is that you would like a module that has a trigger input, and on a trigger generates an output pulse of a given hold-off, duration and  polarity.

Should the output waveform characteristics be defined by generics, or maybe it would be good to have it run-time configurable through an AXI stream interface?

If a new trigger comes while the current output waveform is in progress, what would be the expected behavior?

I am looking forward to hearing your ideas.

tasgomes, 2025-05-29:
@obruendl and @MiloradPetrovic1 

This VC already does that.

https://github.com/VUnit/vunit/blob/master/vunit/vhdl/verification_components/src/std_logic_checker.vhd

You can easily extend it to also expect toggles relatively to the previous one, etc.

obruendl, 2025-08-19:
@MiloradPetrovic1 Is there any progress on this one?