open-logic/open-logic #154: new module development: incremental encoder interface

MiloradPetrovic1, 2025-05-28:
Greetings Oliver!

I think the following module would be a nice addition to open-logic.

A module that takes as input the signals of an incremental encoder (A, B, Z) and outputs the current measured position via an AXI stream.

Some other control and status signals can be useful: strobe, overflow, clear, enable, ...

I can work out the exact detail and write a module interface (ports, generics), if you agree that something like this belongs in open-logic.

I am looking forward to hearing your thoughts!

obruendl, 2025-05-29:
Hi @MiloradPetrovic1 

Yes, sure this belongs into open logic (more precisely into the _intf_ area).

I am looking forward to your contribution!

I suggest the following steps:
1. Read the contributors.md and watch the video about how ti contribute on the open logic youtube channel
2. Fork the repo to your account
3. Add the entity declaration and open a PR, so we can discuss it

From there on we are in touch anyways.

Ok for you?

obruendl, 2025-05-30:
I added my thoughts to the PR.

Are you OK if I close the issue? I think from here on it would be more natural to just discuss in the PR.

MiloradPetrovic1, 2025-05-30:
Yes, you can close the issue. We can continue in the PR.