open-logic/open-logic #159: SPI (or other interface) to AXI-Lite bridge

MiloradPetrovic1, 2025-06-01:
Greetings Oliver.

A few things.

**(1)**

Regarding the SPI to AXI-Lite bridge that is mentioned in the [feature list](https://github.com/open-logic/open-logic/wiki/Feature-Ideas).

Do you already have in mind how the messages from the SPI side should look like? Of course there needs to be a command + address + data, but for example, should there be a checksum for the transmitted SPI data? Also, are you aware of [this repo](https://github.com/airhdl/spi-to-axi-bridge)?

**(2)**

What are your thoughts on other **X to AXI-Lite** bridges? It would be nice to have a real, fast and easy to use memory mapped interface from a MCU/CPU to a FPGA.

Here are some options I think are interesting:
- QSPI to AXI-Lite
- FMC to AXI-Lite, FMC is a general purpose parallel bus used in STM32 microcontrollers.

**(3)**

Also, have you considered maybe **X to AXI-Stream** bridges?

- SPI to AXI-stream (much more easy to implement and use than the AXI-Lite variant)
- PSSI to AXI-stream, PSSI is a parallel interface used in STM32 microcontrollers and processors.
- uPP to  AXI-Lite, uPP is a parallel interface used in Texas Instruments C2000 microcontrollers.

obruendl, 2025-06-03:
Hi Milorad,

Thank you for the repo linked. I didn't know it - I added the linke to the point in the feature list, so I remember to check it and see if there even is a need to cover this in open-logic or if the existing repo is fine.

I did not think about the protocol in detail yet. Generally my mindset was to set-up a "byte-stream to AXI-Lite" protocol that could then be applied to UART as well.

For other busses (QSPI, FMC, and more) I'd not plan to invest any effort unless the community shows a real need and people start to ask for those things.

For the AXI-Stream bridges: The _olo_intf_spi_slave_ already has an AXI4-Stream interface, so no need to crate a separate bridge. And again, in general I'd not invest time unless a real need shows up.

Why do I care about UART/SPI to AXI-Lite but not so much about the other options:
- UART/SPI are by far the most commonly used ones and every microcontroller has them
- AXI-Lite is the must-have requirement to configure register banks in your FPGA from a microcontroller
- For significant data streaming and performant busses etween FPGA and CPU I currently see less use because very often if this need arises an SoC is the better solution anyways. But as mentioned, my decision here is not "not doing it" but "listening to the community".

Is there anything in the issue that requires to keep it open?

Generally for "just a quetstion" I'd prefer if you write me on linkedin or the like. GitHub issues for me are the tool to track bugs and the like, so having many issues open for general question obfuscates the issue tracking a bit.

obruendl, 2025-06-17:
@MiloradPetrovic1 

Yes, this e-mail is correct. However, feature discussions better fit into issues so they are properly documented and available on github

I currently have full focus on running CI in syntheiss, so I do not have the time to really think through a protocol. It sounds super easy but it is not if it comes to make it usable over SPI, UART and potentially I2C and bit-error tolerant.

Can we delay this one until CI synthesis is out. In the meantime I think you can focus on the other entities you started working on. OK?

tasgomes, 2025-06-19:
> 
> Do you already have in mind how the messages from the SPI side should look like? Of course there needs to be a command + address + data, but for example, should there be a checksum for the transmitted SPI data? Also, are you aware of [this repo](https://github.com/airhdl/spi-to-axi-bridge)?
> 

I small note on that repo. Unfortunately, that repository is not maintained anymore. It does not mean the block can't be used but for future features or bug-fixing, you will probably need to fork it.

rbrglez, 2025-07-26:
In the near future, I’ll need a UART to AXI-Lite bridge. I plan to implement it for my own use, but I can also develop it here so that other users can benefit from it as well.

Using the UART to AXI-Lite bridge, I want to be able to initiate AXI-Lite write and read transactions via commands received over UART.

@obruendl, did you think about the protocol that can be used in this implementation?

obruendl, 2025-08-19:
@rbrglez 

I have a protocol in mind. Let me write that down, you can then comment.

The general idea is to assume any kind of byte-stream as medium for the protocol (could be SPI, I2C, UART or anything else).

## REQUEST

**byte1[1:0]** 
Sync bytes, a fixed value of 0x7A5A. Thiese bytes are used to detect the possible start of a packet if alignment was lost due to bit-errors. I propose 16 bits because detecting a sync pattern by accident may significantly slow down resynchronization after bit-errors (which are possible on UART). With 16 bits the chance is 1/64k instead of only 1/256.

**byte[2]** 
Command byte. 
Bit[7]: 1=Read 0=Write
Bit[5:4] number of address bytes in the form 2^N (1, 2, 4, 8 bytes possible)
Bit[1:0] number of data bytes in the form 2^N (1-8 bytes possible) 
Bits [6] and [3:2] ar reserved for future use, potentially to extend the number of data-bytes and also suport AXI4 burst transactions. 

**byte[3]**
ID - a counter increasing with every packet. Used to identify the correct respons to a request (e.g. in case of lost packets). 

Then the address bytes follow, then the data bytes follow.

At the end, a 16-bit CRC for the whole packet (including header, address, data but _NOT_ including the sync bytes) is added to check validity. 

The receiver does ingore packets with non-matching CRCs. In case of errors in the header, synchronization may be lost completely and the interpretation of the next packet is started at the synchronization bytes. If the synchornization sequence is detected anywhere in the middle of a packet, the CRC is fail and the synchronization repeats.

## Response

**byte[1:0]**
Sync bytes, a fixed value of 0x7B5B. These bytes are used to detect the possible start of a packet if alignment was lost due to bit-errors. I explicitly change the sync sequence so it is easy to see if a packet is a response or a request. 

**byte[2]**
Type 
Bit[7]: 1=Read Response (with data) 0=Write response (without data) 
bit[5:4]=AXI response code
Bit[1:0] number of data bytes in the form 2^N (1-8 bytes possible) for read response (ignored for write response) 

**byte[3]** 
ID of the request to be answered then data follows. 

Then the data bytes follow. (no address bytes of course for the response)

At the end, a 16-bit CRC for the whole packet (including header, data but _NOT_ including the sync bytes) is added to check validity. 

The receiver does ingore packets with non-matching CRCs. In case of errors in the header, synchronization may be lost completely and the interpretation of the next packet is started at the synchronization byte. If the synchornization sequence is detected anywhere in the middle of a packet, the CRC is fail and the synchronization repeats.

obruendl, 2026-01-09:
Hi @rbrglez 

I see, you were active. I try to follow up the messages in order.

**Byte Ordering**

To me personally LSB first seems best. This is easier to extend for different data/address width. Also would this allow to use the existing [width converter](https://github.com/open-logic/open-logic/blob/main/doc/base/olo_base_wconv_n2xn.md) without extending it by a byte ordering generic.

**Architecture**

In order to keep it as lghtweight as possible, I suggest to not go parallel in between but have one entity with two FSMs (one for the request, one for the response) and fill the different AXI4-Lite AR/AW signals as the corresponding bits and byte arrive. Once all of them are there, the Valid/Ready handshaking can be intiated.

**Split into Framing and CRC**

Yes, I think this would be good. Generally such a framing mechanism seems valuable to me. Note that it should allow to operate without sync bytes as well (in case it is used over a lossless channel).

What I am not sure is into which area of open logic this would fit best. The "AXI over bytestream" protocol would mostly fit into AXI, the framing probably better into "intf" (at least as long as here is nothing for protocols or the like). And the entity combining the two really fits into "intf". Do you agree or do you see it differently?

**Draft of the protocol**

Seems OK to me.