open-logic/open-logic #160: Incorporate Scoped Constraints into fusesoc

rbrglez, 2025-06-02:
I've recently been evaluating the CDC modules from Open-Logic using FuseSoC. It would be very helpful if constraints could be integrated directly within FuseSoC. To explore this idea, I’ve created a minimal working example that demonstrates the concept, though there’s definitely room for improvement.

You can check it out here:
https://github.com/rbrglez/open-logic/tree/feature/fusesoc-scoped-constraints

To test the setup, follow these steps:
- clone my branch
```
git clone --branch feature/fusesoc-scoped-constraints https://github.com/rbrglez/open-logic.git
cd open-logic
```
- add fusesoc libraries
```
fusesoc library add src
fusesoc library add doc
```
- run vivado tutorial with fusesoc
```
fusesoc run --target zybo_z7 vivado_tutorial
```

Check out the critical warnings I’m seeing in Vivado. These warnings appear after modifying the `*constraints_amd.tcl` file. Could you take a look and see if there's a way to avoid them?

[image]

---

I've created a test target to evaluate several CDC modules:
`olo_base_cc_reset`, `olo_base_cc_bits`, `olo_base_cc_pulse`.

Would it be possible to add this target to the tutorial files? It would be really helpful to have a clear example demonstrating the proper usage of these CDC modules.

rbrglez, 2025-06-02:
I have tested this with vivado version 2023.2 and 2024.2. Both versions give same results.

If I open vivado tutorial with `scripted_build.tcl` (non-fusesoc) I don't get any critical warnings.

I have compared `scripted_build.tcl` with fusesoc generated "scripted-build" and I have come up with a solution, which doesn't give critical warnings. Check out this branch:

https://github.com/rbrglez/open-logic/tree/test/fusesoc-scoped-constraints

Note that I have removed critical warning only for the base constraints. I will remove critical warnings for intf constraints tomorrow.

I had to add two new files: olo_base_import_sources.tcl and olo_base_scoped_constraints.tcl. I am not happy with these additions, but at least I don't get critical warnings. Do you suggest some better solution?

obruendl, 2025-06-03:
Where are the files _olo_base_import_sources.tcl_ and _olo_base_scoped_constraints.tcl_ you mentioned? I found the constraints core-file but not these two TCL files.

Generally, I think it would be nicer if the _olo_base_ module also imports the scoped constraintes (if this is possible) instead of having a separate module. I am not a FuseSoC pro though - could you investigate if FuseSoC allows to import constraints files selectively for specific tools (so we can import them for Vivado only)?

rbrglez, 2025-06-03:
> Where are the files olo_base_import_sources.tcl and olo_base_scoped_constraints.tcl you mentioned? I found the constraints core-file but not these two TCL files.

They are here: https://github.com/rbrglez/open-logic/tree/test/fusesoc-scoped-constraints/src/base/tcl

> Generally, I think it would be nicer if the olo_base module also imports the scoped constraintes (if this is possible) instead of having a separate module. I am not a FuseSoC pro though - could you [...]

Yes, I will try to have only one core in base directory. I will investigate it further today.

rbrglez, 2025-06-04:
@obruendl 

I tested your setup without success. Scoped constraints aren't applied when using fusesoc. I invested quite some time to make your branch work but without success.

Please take a look at my pull request for scoped constraints. In my branch, scoped constraints are now supported in both the FuseSoC flow and the scripted_build.tcl script.

rbrglez, 2025-06-05:
@obruendl 

I have testing scoped constraints in conjunction with verilog top file (vivado_tutorial.sv) And It looks like scoped constraints are always used as expected. I couldn't replicate this issue: #34

Check out my example of verilog_tutorial.sv with three clock domains (system, fast and slow):
https://github.com/rbrglez/open-logic/tree/test/verilog-scoped-constraints

I was using scripted_builld_sv.tcl and scripted_build_sv_scoped_constraints.tcl to generate the project (I wasn't using fusesoc)

obruendl, 2025-06-12:
@rbrglez 

I can confirm that in 2024.2 scoped constraints work in verilog. I had issues in 2023.1. Probably this was fixed in the meantime. I'll document that they work from 2024.2 onwards.

obruendl, 2025-06-12:
A PR is opened for this (#161 ). From here on the discussion is in the PR, so I close the issue.

closed 2025-06-12 by obruendl