open-logic/open-logic #188: Replace custom strength strippers with VHDL-2008 standard (to_01, to_x01)

rbrglez, 2025-06-29:
In `olo_base_pkg_logic`, there are custom strength stripper functions defined: `to01()` and `to01X()`.

Since Open-Logic has moved to the VHDL-2008 standard, these should be replaced with the standardized equivalents:
- Replace `to01()` with `to_01()`
- Replace `to01X()` with `to_x01()`

obruendl, 2025-06-29:
Ideally you just replace all usages of the proprietary functions by the ieee ones and write into the doc that that the functiobs are deprecated and will be removed. 

I'll do so with the next major release - but I'd like to avoid having another breaking change right after 4.0.0.

Ok?

rbrglez, 2025-06-29:
Closing the issue and moving discussion to the pull request: #189

closed 2025-06-29 by rbrglez