open-logic/open-logic #190: clk_meas clock frequency problem

muscoder, 2025-07-12:
clk measurement module is not compatible with higher clock frequency values like 100 MHz. It can be good if it works with higher frequencies. Could you check this problem?

obruendl, 2025-07-12:
I successfully used the module with frequencies around 250 MHz (on AMD UltraScale+ devices).

Can you describe in dezail what leads you to the conclusion that it is not compatible with higher frewiencies?

Please also provide the following information:
- Target Device
- How you integrated Oprn Logic into your design (woth the script provided by open logic, manually, through FuseSoC)

obruendl, 2025-07-28:
Because there is no feedback ,I close the issue.

@muscoder Feel free to reopen it when you come back to the topic

closed 2025-07-28 by obruendl