open-logic/open-logic #198: DMA for Native RAM interface

rbrglez, 2025-08-03:
At work I needed a Simple DMA connected to a Native RAM interface. I developed a basic DMA for this purpose and I believe it's generic enough to be included in the library.

## Option A:

I suggest that the DMA could reuse many of the interfaces from `olo_axi_master_simple`, but instead of using an AXI interface, it would connect to a Native RAM interface. Check out the diagram below.

[image]

## Option B:

An alternative approach would be to implement an AXI Slave to Native RAM translation layer, allowing `olo_axi_master_simple` to be used without modification. However, this would introduce significant overhead.

obruendl, 2025-08-03:
Id clearly vote for option B. This way it would just be a ram controller and the RAM could be used for anything that has an AXI master interface.

For the full axi slave, I'd suggest to write a generic full axi slave and thrn use this in the RAM controller.

Main question for me is if tge RAM is a generic one. What part are you targetting exactly?

rbrglez, 2025-08-03:
In my example I was using RAM from this library (`olo_base_ram_sp` and `olo_base_ram_tdp`). I was targeting Kintex Ultrascale + device (https://www.en.alinx.com/Product/FPGA-Development-Boards/Kintex-UltraScale-plus/AXKU5.html).
But in the future I mean to target also other boards from other vendors (Altera, Effinix, ...)

If you won't have time for developing full axi slave, then just provide an interface and I can start with the implementation when I'll have time.

obruendl, 2025-08-03:
For AMD you could also just use the "AXI EMC" IP core that is provided by Vivado.

Let me think a bit about the AXI full slave. It is something I have in mind for a while and there are a few non-trivial aspects to it. Feel free to remind me in a week if I did not come back to the topic.

tasgomes, 2025-08-03:
The AXI BRAM Controller from AMD offers a AXI4-Full to native RAM converter.

rbrglez, 2025-08-03:
> The AXI BRAM Controller from AMD offers a AXI4-Full to native RAM converter.

I'll then use AXI BRAM controller until axi slave is developed

obruendl, 2025-08-10:
I renamed this issue because the bigger topic (and first one to tackle) would be a AXI4-Full (burst capable) slave.

Here are my considerations to the topic (I number them, so you can easily reference them in downstream discussions):

1. It's obvious that this component would have a full AXI4-Slave interface
2. IMO we can omit advanced features like the cache signals and the exclusive access control. Executing normal burst read/write is sufficient IMO (should be cleanly documented of course).
3. The Slave shall support all burst modes (incrementing, fixed, wrapping). It _may_ provide a generic to suppress some of them but it _shall_ allow implementations supporting them all.
4. There is no need to support narrow bursts - however, this should be documented.
5. The slave shall allow defining AXI data width and user-logic data-widht separately and convert between those. 
    - Reasoning: It is common for full AXI to have relatively wide busses, still user slaves might have narrower word widths.
6. IMO the component can focus on slaves that do allow random access to memory (SRAM, Register-banks, BRAM, etc.). For more advanced slaves (e.g. DDR memory controllers with their Banks FSMs) I think it is okay to say "write your own". Hence, breaking down transfers to "Address/Data" pairs seems OK to me.
7. I asked myself if the component should include clock crossings (between AXI and user-side) but I concluded that it shall not - but have an interface that allows the user to attach them externally.
8. Interface for write transactions should be as follows IMO:
    - Wr_Valid 
    - Wr_Ready (so the user logic can throttle speed if required, e.g. if a CDC fifo is required)
    - Wr_Addr
    - Wr_Data
    - Wr_Be
    - Wr_Last (Last beat of a burst)
    - Wr_First (First beat of a burst)
    - Most of this is obvious. _Wr_Last_ and _Wr_First_ are meant for any user logic that is burst optimized.
9. The AXI4-Full slave shall _NOT_ include a write data FIFO. Reasoning: 
    - This can be attached to the _Wr_xxx_ interface easily if required
    - Not including the FIFO does allow the user to choose a sync or async FIFO
    - Connecting the FIFO externally does allow optimization (e.g. not include _Wr_Addr_ in the FIFO data signals to save RAM)
    - Because for write transaction the timing of the _W_ channel is controlled by the master, integrating a FIFO would not lead to any advantage.
10. Interface for read transactions should be as follows IMO:
    - RdAddr_Valid
    - RdAddr_Ready (so the user logic can throttle speed if required, e.g. if a CDC fifo is required)
    - RdAddr_Addr
    - RdAddr_Last (First beat of a burst)
    - RdAddr_First (First beat of a burst)
    - RdResp_Valid
    - RdResp_Ready
    - RdResp_Data
11. The RdResp channel shall operate in either "low-latency" or "normal" mode. In "normal" mode the complete read response is buffered and then sent to the AXI RRESP channel as one burst to limit the time the channel is used. In low latency mode, data is sent to RRESP as soon as possible (see AXI masters for details). 
12. Due to (11) a FIFO is required on the _RdResp_ path.
13. For testing the existing _olo_test_axi_master_vc_ may be used (but this is not mandatory of course as long as test coverage is OK).
14. The slave shall allow automatic ID propagation (i.e. the AW_TID is propagated to the the B_TID automatically).