open-logic/open-logic #202: Unsafe path on LUTRAM when using async FIFO or olo_base_ram_sdp.vhd

tasgomes, 2025-08-08:
I am using the `olo_base_fifo_async` entity in Vivado 2024.2 as follows:

```vhdl
    i_fifo : entity work.olo_base_fifo_async
    generic map(
        Width_g                         => 27,
        Depth_g                         => 64,
        AlmFullOn_g                     => false,
        AlmFullLevel_g                  => 0,
        AlmEmptyOn_g                    => false,
        AlmEmptyLevel_g                 => 0,
        RamStyle_g                      => "auto",
        RamBehavior_g                   => "RBW",
        ReadyRstState_g                 => '1',
        Optimization_g                  => "SPEED",
        SyncStages_g                    => 2
    )
```

Due to the small size of it, the RAM style ends up being a LUTRAM:

[image]

In this scenario, I realized that the inter-clock path between these two clock domains was not being constrained with an exception, i.e., Vivado was analyzing them as a timed path. Depending on the clock source, their frequencies and phases, the timing on this path can fail.

I now added a scoped-constrained for this path (a merge-request will follow). 

However, as described in [UG906](https://docs.amd.com/r/en-US/ug906-vivado-design-analysis/LUTRAM-Read/Write-Potential-Collision), this path could still be unsafe if there is a read and write access to the same address at the same time. 

So unless there is logic that prevents any read/write collision I think it is unsafe to use this block (in LUTRAM mode). 

At the end, I am not sure if adding a scoped constraint is recommend as this will hide a potential dangerous path.

I don't think the current implement prevents a read/write collision, right? What would you like to do?

- Add read/write collision logic + scoped constraint
- Not allow `RamStyle_g` to be `auto`, so that we force it to be BRAM or URAM, etc.

tasgomes, 2025-08-11:
An async FIFO never allows a valid read and write to the same address at the same time, by design. 

Hence, this path can be safely ignored from timing analysis using a scoped constraint when the entity `olo_base_ram_sdp.vhd` is used in the context of an async FIFO.

However, when this entity is used as standalone, it is probably not a good idea to apply automatic scoped constraints on this path. The designer might not be aware that this path could be unsafe if there is a read and write access to the same address at the same time. It is better to let Vivado warn the user about it.

Hence, I propose we only add a scope constraint for the async FIFO module.

obruendl, 2025-08-15:
PR merged, hence I close this issue. @tasgomes Feel free to reopen if you disagree

closed 2025-08-15 by obruendl