open-logic/open-logic #206: Hashtable

AnthoJack, 2025-08-17:
Hi,

I'm currently working on a project at work that required a hashtable so we implemented one that supports read, write, remove, clear and key output (to get the keys stored in the table). AFAIK there isn't any in open-logic currently so if there is any interest in one, I've been allowed to provide ours. It would require a bit of work to adapt it to open-logic's framework and pipeline but I'd gladly do so if you're interested. If so, I can start working on the documentation to give you more informations on how it works. Then, you can give some feedback if you'd like some modifications before starting to work on the actual re-implementation

Have a nice day

obruendl, 2025-08-19:
The code seems pretty fine. I did not yet review every line but I walked through a few scenarios and for those everything looks OK. So **yes, please proceed**.

In the documentation, please ensure the cluster removal logic is documented well in the _archtecture_ section - this is the only part I did not understand from the code easily. Of course the general architecture shall be documented as well (for users) but the cluster removal logic I also need documented to understand it myself.

A few other notes:
- We should think of adding at least oen or two more advanced hash functions than just picking the lowest bits. I am not a master of hash functions - so I cannot help much. Open Logic has a CRC (I think this could be used as simple hash) but there probably are better candidates.
- I see that you are always using `reg_v` - when the code is adapted for open logic, `reg` should be used wherever possible in conditions and on the right-hand side of assignments (so one can spot longer combinatorial paths easily).
- The RAM of course should be replaced by _olo_base_ram_sdp_ instances.

AnthoJack, 2025-09-01:
I made a WIP PR with the documentation I've created. I described the hashtable as it WILL be implemented, not as it is in the code I submitted here since it will have to undergo significant adaptations

obruendl, 2025-09-27:
I close the Issue because the PR is open and we gonna have all other discussions there.